andrewshilliday/garage-door-controller #6: Feedback

HarleyGolfGuy, 2015-02-09:
Is there a place in github for discussion regarding a project?  I don't see one, so here goes:

* First off, great project, software worked right out of the box.  
* This was my bad, but maybe the links to the actual products should be omitted because things change so rapidly...for example the raspberry pi link is broken so I found one easily on Amazon, then I bought the 8GB SD card from your link, but my raspberry pi b+ uses a microSD card.  So I had to return via Amazon...now I know that was mostly my fault but...
  * There is a slight glitch in the software, when the garage door is closing sometimes it will say "open" for a while.  
* I'm still trying to find out how to wire the output from the relay to the garage door opener properly.  When I wire it, it works but then for some reason my opener on my wall that we use all the time doesn't work.  All other remotes, even the sensor work but just not the main switch.  If you have any ideas please let me know.

Overall well done!  I'm up and running!  Very cool.

budpi, 2015-02-13:
HarleyGolfGuy

I ran into the same issue with the wall button not working. I just switched the wires on the back of the wall button and all was well. Hope this helps or that you figured this out by yourself...

By the way, I too wish to get rid of the password intigration (to the extent that I do not wish to enter it everytime) You commented that you wanted to eliminate it all together...any luck?

thanks,

budpi

cali0028, 2015-02-26:
Great job on this project!

After searching and reading this is the only that I actually liked that was simple and clean! 

Just did the s/w side of it on my pi but have not installed the last peace...hooking it up to my garage! That will happen this weekend. 

One thing I couldn't figure out is how do I remove the second user / door? I only have one door and two users. 

The password requirement I got away by using chrome on the iOS and Android. 

I've also managed to run this using freedns using duckdns.org if anyone is interested. They have a Rpi package that you can install so the IP get updated for changes every 5 min. 

Wish List: Webcam support. Looking at the UI I think there is enough space to include a webcam feed.

jsmithfraymangroup, 2015-09-06:
Hi thank you for this project.  I dont have the door sensor hooked up yet (parts on the way). but have wired up the pi to the relay and the door opener.  The door does not open/close.  I was wondering if the missing sensor would prevent the door from opening/closing?

andrewshilliday, 2015-10-05:
jsmith, the sensors shouldn't be needed.   try taking the wires coming from your garage door controller and touching them together.  Does the garage door open?  If not, then your controller is either miswired, or is using a protocol over that wire that is more than just a "close the circuit to open the garage door" kind of thing.

andrewshilliday, 2015-10-05:
HarleyGolfGuy,   I think that you can disable authentication by removing it from the settings file but my official answer is "try it and see what happens" :)

ronlh1, 2015-10-19:
I see there are lines where I can add send to email. How do I configure and do I have to install any other software to have it send email notification?

Thanks,
Ron

kfakhri1, 2015-11-04:
Just finished installing this. Thanks for putting this up!  Few questions still remain, first, has anyone figured out how to remove/modify the login/password requirement?  I modified the configure.json file, but it disabled the whole software, so I changed it back.  I read above that there's a setting in the twisted Python library.  Where would I find the library on my Pi?  Second, any luck with the webcam integration.  I've installed mine without the sensors, so a webcam would be my way of checking on the door.   Third, any instructions on how to set email notifications? In case I decide to install the sensors.  Thanks again.

0-kaladin, 2015-11-04:
@kfakhri1 I made the web authentication an option if you want to make the same changes: https://github.com/0-kaladin/garage-door-controller2/commit/d517c1ded5497844bed9defd460fdd292c4652a3

then in your config.json in the options block just add
 "auth":"False",

this will allow you to easily enable again if you ever want it.

kfakhri1, 2015-11-06:
0-kaladin, I don't have an options block in my config.json file.  Where would I enter the "auth":"False", code?  Thanks for your help.

jtonzi, 2015-11-06:
@kfakhri1 , you need to add one. 
I put this in my config.json (in between the "site" and "doors" blocks, but I don't think it matters where)

"options" :
        {
        "auth":"True"
        },

kfakhri1, 2015-11-06:
jtonzi,  Thanks. That worked!  Next step, integrating a webcam feed into the code!  Thanks again.

Dk-guru, 2016-03-08:
Andrew, 

Thanks for the project! I used this for a commercial grade opener; it works perfectly for each function: Stop, Open, Close.  
I did have to use http://www.amazon.com/SainSmart-4-CH-4-Channel-Relay-Module/dp/B0057OC5O8 with the Raspbery Pi2 Model B+ and to protect against power failure: http://www.hellonomad.com/products/nomadplus.

Daniel

budpi, 2016-03-27:
I successfully integrated a webcam by following these instructions for installing Motion from this website.

[http://stuffbabiesneed.com/blog/building-raspberry-pi-baby-monitor-part-one/](url)

_I've used this software for other projects and it works just fine._

HARDWARE USED:

* Playstation EyeToy Webcam *

After you have the program installed, EDIT your 

/home/pi/garage-door-controller/www/index/html file

to include this code somewhere between the <body></body>

[image]

SPECIAL NOTE!!!

Be sure to change the port 8081 to something like 8083 in this file:

sudo nano /etc/motion/motion.conf 

(Not sure how important this last part is but, I figured that using the same port as the Garage-Door-Controller may cause some conflict.

--------------------------- 

The Webcam can also be ran from other Raspberry Pi and this code also works for other IP Webcam software/apps, like ones you download from the PlayStore and run on an old Android Phone.

Hope this helps anyone interested. If there are other, easier ways that work better. Please, do share.

rapi3, 2016-04-18:
hello,

I setup this project on Raspberry Pi (_ thank you for your hard work to code_ ) and now I am testing on my desk with 8 board relays.
I am not using contacts for real state close/open and I assigned for all doors unused "state_pin" : 26.
Commands for all 8 relays works ok,

 **only problem I have is with email sending:**
I tried to test it with gmail and yahoo accounts as server on both ports 465 and 587 but no luck to send emails from pi.
in log I see: **action 'action 17' suspended**

_**Is it email part broken in last code ? Or is something wrong in my set-up ?**
Any idea ?_

thank you very much.

p.s
what is openhab for ?
"openhab" 
```
	"server" : "<SERVER_IP_ADDRESS>",
	"port" : "<OPENHAB_SERVER_PORT>"
```

part of config:

```
.......
	"use_auth":true,
	"use_alerts":true,
	"use_openhab":false
},
"alerts":
{
	"time_to_wait" : 10,
	"alert_type" : "smtp",
	"smtp":
		{
		"smtphost" : "smtp.mail.yahoo.com",
		"smtpport" : 465,
		"smtp_tls" : "True",
		"username" : "yyyyyyyyy@yahoo.com",
		"password" : "xxxxxxxxxxxxxxx",
		"to_email" : "zzzzzzzz@gmail.com"

.............
"doors" :
	{
		"releu1" :
			{
			"name" : "Usa 1 deschide",
			"relay_pin" : 17,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu1"
			},
		"releu2" :
			{
			"name" : "Usa 2 inchide",
			"relay_pin" : 18,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu2"
			},
		"releu3" :
			{
			"name" : "Usa 2 deschide",
			"relay_pin" : 27,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu3"
			},
		"releu4" :
			{
			"name" : "Usa 3 inchide",
			"relay_pin" : 22,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu4"
			},
		"releu5" :
			{
			"name" : "Usa 3 deschide",
			"relay_pin" : 23,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu5"
			},
		"releu6" :
			{
			"name" : "Usa 4 inchide",
			"relay_pin" : 24,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu6"
			},
		"releu7" :
			{
			"name" : "Usa 4 deschide",
			"relay_pin" : 25,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu7"
			},
		"releu8" :
			{
			"name" : "Usa 1 inchide",
			"relay_pin" : 4,
			"state_pin" : 26,
			"approx_time_to_close" : 20,
			"approx_time_to_open" : 30,
			"openhab_name" : "g_door_releu8"
			}
```

part of log

```
**Apr 18 15:15:54 rpi3 garage_controller: we are using SMTP**
....
> garage_controller: Usa 2 inchide: closing => open
> Apr 18 15:16:26 rpi3 garage_controller: Usa 1 inchide: closing => open
> Apr 18 15:16:30 rpi3 garage_controller: Usa 1 deschide: closing => open
> Apr 18 15:16:31 rpi3 garage_controller: Usa 2 inchide: closing => open
> Apr 18 15:16:33 rpi3 garage_controller: Usa 2 deschide: closing => open
**> Apr 18 15:17:01 rpi3 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Apr 18 15:18:01 2016 [try http://www.rsyslog.com/e/2007 ]**
> Apr 18 15:17:39 rpi3 garage_controller: Usa 2 deschide: toggled
> Apr 18 15:17:40 rpi3 garage_controller: Usa 2 deschide: open => closing
> Apr 18 15:17:49 rpi3 garage_controller: Usa 2 deschide: closing => open
> Apr 18 15:18:32 rpi3 garage_controller: Usa 2 inchide: toggled
**> Apr 18 15:18:32 rpi3 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Apr 18 15:19:32 2016 [try http://www.rsyslog.com/e/2007 ]**
> Apr 18 15:18:33 rpi3 garage_controller: Usa 2 inchide: open => closing
> Apr 18 15:18:42 rpi3 garage_controller: Usa 2 inchide: closing => open
**> Apr 18 15:23:58 rpi3 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Apr 18 15:24:58 2016 [try http://www.rsyslog.com/e/2007 ]**
...

```

crispyjones, 2017-07-07:
Thanks for this, wanted to pass along my experience.

Built without issue using Raspbian Jessie Lite on Rpi Zero W, everything fits on a 2gb card.

I built my own transistor driven relay and had to reverse the logic of the relay pin, simply flipped True/False in controller.py in the few places gpio.output(self.relay_pin xxxx) occurred

ronlh1, 2017-07-08:
I have a problem sending email as well
Anyone have a solution?

crispyjones, 2017-07-08:
I had a couple issues with out of the box config using gmail.

First, in config.json change the alert type line to the following
"alert_type" : "smtp",
Then fill in the rest of the variables with your account info, port 587 worked for me.

On my first attempt my gmail account alerted me to an insecure login attempt, I had to go to settings and allow insecure devices for it to function, the instructions for changing the settings are in the bounce email. I did this on an old gmail account that I consider throw away, I would *not* do this to my "real" account as the password is saved in plaintext on the pi and you are degrading the overall security of the linked gmail account.

Maybe some linux gurus have a more secure way to do this, but this is how I got it up and running.

ronlh1, 2017-07-10:
I'm still having issues with sending email. Here is a copy of my config.json

Dk-guru, 2017-07-10:
I may be using an older version... but I would remove the stuff that you do not use: i.e. 
remove pushbullet, access_token, pushover, user_key, and api_key.

Here is what my config.jason file has:
```
{

"smtp":
	{
	"smtphost" : "smtp.gmail.com",
	"smtpport" : 587,
	"smtp_tls" : "True",
	"username" : "garage.door@.com",
	"password" : "<AG",
	"to_email" : "d@.com",
	"time_to_wait" : 1740
	},
"site" :
	{
	"port":881,
	"username":"f",
	"password":"1"
	},
"doors" : 
	{
		"left" : 
		       {
		       "name" : "CLOSE",
		       "relay_pin" : 23,
		       "state_pin" : 27,
		       "approx_time_to_close" : 15,
		       "approx_time_to_open" : 15
		       },
		"middle" : 
		       {
		       "name" : "STOP",
		       "relay_pin" : 22,
		       "state_pin" : 27,
		       "approx_time_to_close" : 2,
		       "approx_time_to_open" : 2
		       },
		"right" : 
			{
			"name" : "OPEN",
			"relay_pin" : 24,
			"state_pin" : 27,
			"approx_time_to_close" : 15,
			"approx_time_to_open" : 15
			}
	}

}

```

jtonzi, 2017-07-10:
I'm with Dk-guru on this one.  Remove the stuff you don't need.  Also, I have the time_to_wait variable inside the smtp section.  It may be needed.

ronlh1, 2017-07-10:
I have the time to wait in the smtp section, DK-guru has the time to wait
under to email.
If I change the alert type from null to smtp the web page will not even
load.

jtonzi, 2017-07-10:
I think you're going to need someone else to weigh in. Looks like Dk-guru and I are on an older version (I haven't updated in well over a year). I wish I could help. Best of luck.

andrewshilliday, 2017-07-11:
Are you seeing an error printed to the console, or are you simply not receiving emails?

ronlh1, 2017-07-11:
I'm not receiving emails. I have tried different email accounts as well.

andrewshilliday, 2017-07-11:
Were they all gmail accounts?  Try running the program manually from the command line rather than automatically at startup.  That should make it so that it prints any errors to the screen.

ronlh1, 2017-07-11:
I have tried outlook email as well. how do I run manually?

Thanks

sqlcursor, 2017-10-16:
Thanks for the tutorial.  This is great and exactly what I needed.  The only headbanging moment was when getting the right door sensor to correctly report the state of the door.  When I looked at the pin diagram on your site, you have the right door sensor going to what is labeled GPIO21.  When I put that in the config.json file it did not work.  It wasn't until I looked at an alternate pin diagram that I discovered it was actually GPIO27, not GPIO21.  When I changed the config.json file with that, the sensor started reporting the state appropriately for that door.  Again, I can't thank you enough for this and it's working flawlessly!

closed 2018-02-15 by andrewshilliday