andrewshilliday/garage-door-controller #7: Error while starting

Hondo88, 2015-02-21:
Thanks for sharing your work on this, its exactly what I've been looking for. I've tried to get the application going and ran into some trouble. Everything seems to install well, but after running the controller.py I get the following errors. 

$ python controller.py
  File "controller.py", line 279, in <module>
    controller = Controller(json.load(config_file))
  File "/usr/lib/python2.7/json/__init__.py", line 278, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 326, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 365, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 381, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting , delimiter: line 10 column 2 (char 189)

The install seems pretty strait forward so I'm not sure if I'm doing something wrong. 
I don't have any experience in code writing, so I'm not sure how to troubleshoot this.

Meatwad, 2015-02-22:
There is a typo in the config.json file.
Line 9 should have a comma after the "<TO EMAIL ADDRESS>" entry.  Once I put the comma in place it works fine.

Hondo88, 2015-02-25:
Thank you very much! Works great!

closed 2015-02-25 by Hondo88