andrewshilliday/garage-door-controller #8: relay closes within 2 secs

Brian001, 2015-07-23:
Hi there,

Awesome project but i have a small issue: 

when i want to Open the closed garage door , i hear the relay clicking and it clicks again within a sec!
same goes for the other relay and the state of the door doesnt matter (open or closed)

Hope that someone here can help

Thank you in advance

Brian.

andrewshilliday, 2015-10-05:
This is the intended behavior.  The relay's purpose is to simulate a human pressing the button on the wall.  The human doesn't hold the button down the entire time the garage door is opening/closing, they just push the button and release while the door moves.  The two clicking noises on the relay are the "push" and "release" actions and should occur within about a second of each other.

rwadeg, 2016-08-07:
My 25+ year old garage door opener requires at 2 to 3 second press for it to activate.
I was able to keep the relay activated (or closed) longer by changing a line in the "controller.py" file.
Here's how:
open the file controller.py with a text editor (/home/pi/garage-door-controller/controller.py)
find the line that says:  time.sleep(0.2)
change the: (0.2) to (3.0) i.e 3.0 seconds
save the file and restart program or reboot device
now when you push the garage door button the relay will hold closed for 3 seconds

andrewshilliday, 2017-01-20:
Sorry for the delay in responding.  rwadeg is correct about the code to change.  If there is interest, I could put that variable into the config.json file.

closed 2017-03-04 by andrewshilliday