andrewshilliday/garage-door-controller #10: Garage door operates intermittently

Joeybuttta, 2015-08-08:
Thanks for a well explained and easy to do project. I have everything up and running, with one issue. When I access the webpage and press the screen to have the garage door operate, the relay always click but it doesn't always operate the garage door. I usually have to press it 3 or 4 times before the garage door moves. I have a CanaKit 5V 2.5a Raspberry Pi 2 Power Supply and I have the relay powered as instructed from GPIO 5V pin (i even tried both of the 5V GPIO pins). I checked all wiring connections and everything seems connected. Any suggestions on what the issue could be? It sounds like it's not passing enough volts to me, but I'm no expert.

Any help is appreciated. 

Thanks,
Joe

andrewshilliday, 2015-10-05:
This is strange.  the RPI doesn't pass any current to the garage door controller, it just closes a circuit build into the controller.  I wonder if perhaps the relay is not closing the circuit for long enough time.   If you're good at python, you should be able to find the chunk of code that determines how long the relay is clicked for and adjust it.  If not I can point you to the spot in the code.

Joeybuttta, 2015-10-05:
I don't particularly know python but I believe this is where that control is:
  gpio.output(self.relay_pin, False)
  time.sleep(0.2)
  gpio.output(self.relay_pin, True)

I'll make that change and try it again. Thanks for the response.

andrewshilliday, 2016-01-03:
Yup, that's the right spot.

Joeybuttta, 2016-01-14:
I must have corrupted my SD card. RPi still works to open the door, but won't save any changes when I  update the file and reboot. I want to re-install everything on a new SD card but haven't had time to do that.

Joeybuttta, 2016-01-17:
I changed time.sleep to (0.4) and the garage door activates the first time, so that did the trick. Thanks again.

Have you incorporated a Raspberry Pi camera into your setup. I'm going to attempt that over the next month or so, but don't want to mess with the current setup you already did.

any suggestions on how to get that working?

andrewshilliday, 2017-01-20:
One year late:  No idea how to integrate a camera.  If you got that working, please let me know.

closed 2017-01-20 by andrewshilliday