andrewshilliday/garage-door-controller #11: send_opendoor_message fails to send mail and so status_check stops working.

EPIC3D, 2015-08-13:
The garage status readout stops working.

It happens when the "time_to_wait" : 1800 happens and it tries to send mail.

In the syslog the message arrives about sending mail.

Aug 13 10:02:13 raspberrypi garage_controller: Sending open door message. (1800)

So it reaches this part:
syslog.syslog("Sending open door message. (%s)" % opentime)
But then after it, the status readout stops working and also no mail sent at all.

When I turned off this part the satus readout was working fine after (ofc.. still no mail since I switch it off)

This is what "fixed" the isssue for me. But I would like to have a working mail send too. It is just a temporary solution for me.

def send_opendoor_message(self, opentime):
        syslog.syslog("Sending open door message. (%s)" % opentime)
        config = self.config['smtp']
        # server = smtplib.SMTP(config["smtphost"], config["smtpport"])
        #if (config["smtp_tls"] == "True") :
        #    server.starttls()
        #server.login(config["username"], config["password"])
        #message = "Your garage doors have been open for %s." % elapsed_time(100+opentime)
        #server.sendmail(config["username"], config["to_email"], message)
        #server.close()
        self.msg_sent = True    

I'm no programer. So I hope this report helps.

andrewshilliday, 2015-10-05:
Hm.  Are you certain that your email settings are configured correctly?

EPIC3D, 2016-02-16:
Sorry for the very late answer.

Yes. I think I had the settings right. I also tried different e-mail accounts and servers. No luck.

Still. Even if i would set it wrong or the mail server not working, maybe the status readout shoud not stop working in case of an error.

Currently I use my own garage door control program. It is very simple and not as sophisticated as yours. Only have an open and a close command. No sensors. I can only check the door from a camera feed.

So I would like to use your program in the future since it have a very neat status readout, display and potential e-mail alarm system.

Gilles94500, 2016-02-26:
I have exactly the same issue. The message is written in the syslog, then nothing happens. The webserver doesn't anwer anymore , the mail is not sent. I have doublechecked the mails parameters, everything looks correct. I am using a email account @laposte.net.
Each time I have to stop/start controller.py or I reboot the raspberry.
Any clue?
Thanks
Gilles

lamping7, 2016-02-27:
The code should really handle exceptions, then you will will have a better idea of what is going on, if an error exists.

Try this code instead and let us know how it turns out. If an exception is thrown, it should show up in the syslog. Disclaimer: I did not test it. 
```python
    def send_opendoor_message(self, opentime):
        syslog.syslog("Sending open door message. (%s)" % opentime)
        config = self.config['smtp']
        message = "Your garage doors have been open for %s." % elapsed_time(100+opentime)
        try:
            server = smtplib.SMTP(config["smtphost"], config["smtpport"])
            if (config["smtp_tls"] == "True") :
                server.starttls()
            server.login(config["username"], config["password"])
            server.sendmail(config["username"], config["to_email"], message)
            self.msg_sent = True
        except smtplib.SMTPException as e:
            syslog.syslog(str(e))
        server.quit()
```

Gilles94500, 2016-02-27:
Hello,

Thanks for the prompt answer.

I have applied the below config, but I didn’t get anything as it freezes and never gives the hand back.

Here is the syslog( I have setup 60 sec instead of 1800 sec)

 

Feb 27 15:08:41 raspberrypi garage_controller: Garage: closed => opening

Feb 27 15:08:50 raspberrypi garage_controller: Garage: opening => open

Feb 27 15:09:41 raspberrypi garage_controller: Sending open door message. (60) 

 

I also did some debug(import ipdb; ipdb.set_trace())  but didn’t see anything.

I also have added a timeout(that I have removed for your test):

        server = smtplib.SMTP(config["smtphost"], config["smtpport"],timeout=30) so it fails with timeout reached and continue to run, but the mail is not sent.

I have changed the port (587 instead of 465) and I receive a reject message from the mail server, so the communication exists.

I have done some other tests with another email yahoo.fr instead of laposte.net : same issue.

What is surprising me is that on the same raspberry, I use mutt with motion and I have no issue with the same email/server/port.

 

It is like if the mailserver or controller.py  was waiting for something which never happens.

 

Thanks for your help

 

Regards,

Gilles

lamping7, 2016-03-01:
What was the reject message? 

Do you have "smtp_tls" : "True" in config.json?

You can also give this a try:
server = smtplib.SMTP_SSL(config["smtphost"], config["smtpport"])

Gilles94500, 2016-03-01:
What was the reject message? No reject message, it freezes until the timeout expires.

Do you have "smtp_tls" : "True" in config.json? YES

You can also give this a try:
server = smtplib.SMTP_SSL(config["smtphost"], config["smtpport"]) I also did it (need to remove the start_tls()) . I get the error message :”service unavailable , please retry later” .

I also tried to add a Subject: I get frozen as before.

I don’t know a lot about Python and don’t know what t debug. It seems that the smtp server is waiting for something wihich should come along with the server.sendmail statement.  I have googled it with no success. 

Here is a part of the config.json file.

{

"smtp":

        {

        "smtphost" : "smtp.laposte.net",

        "smtpport" : 465,

        "smtp_tls" : "True",

        "username" : "sender@laposte.net",

        "password" : "sender_pwd",

        "to_email" : "recipient@laposte.net",

        "time_to_wait" : 60

        },

 

Regards,

Gilles

Gilles94500, 2016-03-01:
Got it. 

I have just changed the port from 465 to 587 an now it works. I didn’t notice that Mutt is using 465 with smtps (not smtp);

Btw, I have had to add a sender( From: ) else the mail is automatically moved to the junk mails directory by the mail server, so no chance to get it in Outlook.

 

Thanks for your time and patience.

 

Regards,

Gilles

andrewshilliday, 2017-01-20:
You guys are absolutely right that the code needs better error handling.  I'm not a python expert so I didn't do a good job of that.  If someone wants to clean up the error handling, please issue a merge request and I'll happily integrate it back into the master.

closed 2018-02-15 by andrewshilliday