andrewshilliday/garage-door-controller #12: Relay activating on on startup.

SouthernPlayer, 2015-10-20:
My relay was activated on startup. I was causing my garage to open as soon as the RaspberryPI came online. Not good when there is a power outage. I had to change the following in controller.py.

42   gpio.output(self.relay_pin, False) <--was True

72   gpio.output(self.relay_pin, True) <--was False
73   time.sleep(0.2)
74   gpio.output(self.relay_pin, False) <--was True

jtonzi, 2015-11-02:
I had the same issue too.  I first started using GPIO14, which is apparently HIGH on startup (causing the garage to trigger after power on).  I switched to GPIO23 and it was fine.  I found this list of GPIO states on startup:  https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=35321

I think if you pick anything with a LOW on startup it should be fine.

SouthernPlayer, 2015-11-02:
That is strange, because I did not deviate from the readme. I am using GPIO23.

jtonzi, 2015-11-02:
Maybe a different revision of the Raspberry Pi? I'm using the old old rev. 1 model B.  Not sure.  Good to hear you fixed it though.  
When I first started setting it up and the relay was triggering on startup, I almost gave up on the project (don't need my garage opening randomly in the night).

andrewshilliday, 2016-01-03:
I think there's some sort of configuration on the RPI that controls how those pins are supposed to be set at startup.  They might be configured differently for different people and that might be causing the problem.

lamping7, 2016-02-17:
Different RPI models have different startup states, as pointed above. Some are also known to be in floating states. I think there is confusion on how this circuit works. The relay board is active LOW. If you look at the toggle_relay function in the Door class in controller.py you'll see:
`       gpio.output(self.relay_pin, False)
        time.sleep(0.2)
        gpio.output(self.relay_pin, True)`
The relay triggers when the GPIO goes LOW (0/False/0v)), then holds that state for 0.2 seconds, and sets back to HIGH (1/True/3.3v).

If you google this issue, you'll see some people try different GPIO pins (because it isn't thought to be a big deal to them), while others make the circuit more robust and use a ULN2003 IC driver chip (consisting of the necessary transistors, diodes, and resistors). The IC chip will also switch the relay to active high from active low, so there is some minor code modification required in the function shown above.

morejava, 2016-05-19:
I had the same issue.  If I rebooted my router or pi the relay would activate either closing or opening garage.  BIG security issue.  Strange thing was that I used a JBtek relay with no issue.  Never switched the GPIO wiring or code.  The relay that didn't work was some cheap relay I got in a module pack from amazon.

hjone72, 2017-09-26:
I found that by editing line 50 of controller.py from:

`gpio.setup(self.relay_pin, gpio.OUT)`

to:

`gpio.setup(self.relay_pin, gpio.OUT, initial=True)`

resolved the issue.

6jacksons, 2017-11-30:
I am having a similar issue, where upon running the app, the relay is energised, and only a reboot of the Pi Zero W de-energises it. I’ve tried adding the ‘initial’ setting, changing the true and false settings, no change. It’s configured to use GPIO23.
Any suggestions?

Cirrostratuss, 2018-01-02:
What is the path to controller.py?

6jacksons, 2018-01-02:
/home/pi/garage_door_coltrollerv2/