andrewshilliday/garage-door-controller #13: HTTP GET?

jtonzi, 2015-11-05:
Not really an issue.
Is there a way to trigger the doors using an HTTP GET? It would be nice to open a bookmark in a browser and the door immediately trigger.

Thanks!

p1neapples, 2016-01-14:
+1 to this. My use would be via my Pebble, where I could just press a HTTP request button when I get on my street that would kick off the open.

jtonzi, 2016-01-14:
That's exactly what I'm hoping to do with it as well.  I'll let you know if I'm able to sort it out.

Pack3tL0ss, 2016-01-14:
You can operate the button (simulate clicking the door in the web interface) by browsing to 
http://<IP of Controller>:8081/clk?id=<door id>

Where 'door id' is in the spot where "left" is on my controller as show below:
"doors" : 
        {
                "left" : 
                       {
                       "name" : "Garage",
                       "relay_pin" : 23,
                       "state_pin" : 17,
                       "approx_time_to_close" : 16,
                       "approx_time_to_open" : 14,
                       "relay_hold" : 0.5
                       }
        }

So I have a link on my mobile device home-screen that looks like
http://garagepi.home.lab:8081/clk?id=left

It'll simply return 'OK' in the browser, but it operates the relay and opens/closes the door.

jtonzi, 2016-01-14:
will that still work with the authentication?  If so, how would I pass the username and password?
Thanks!

Pack3tL0ss, 2016-01-14:
I have authentication enabled, but I have it saved in my mobile browser.  I prefer it enabled... it's kind of a confirmation that I actually meant to hit the button, but when it pops up the auth window my credentials are already there (had my mobile browser save them), so I just have to hit "Login" and the door operates.  

I'm not sure there is a way to pass the authentication in the http request, although it is possible to disable it (wouldn't be ideal if you've exposed the controller outside your network).

jtonzi, 2016-01-14:
I think that's the problem.  Ideally I'd be able to use my pebble to hit the site (your method would be great, but like you said, mine is reachable outside and leaving it unprotected would be bad).
Thanks for your help though, being able to simulate the click is a good start.

p1neapples, 2016-01-16:
Figured it out. To pass username and password thru a GET request from my Pebble (or anything) here is the schema:

http://username:password@webaddress.com:8081/clk?id=left

Working like a charm today. Hope that works for you too.

jtonzi, 2016-01-16:
Worked! Thank you!

http://username:password@webaddress.com:8081/clk?id=left

budpi, 2016-02-10:
Awesome!!! Thank you so much!!!

budpi, 2016-03-16:
How would you make the command return a different page/link besides just showing "OK"? Is this something that can be done?

shiznic, 2016-04-27:
This works great!  Any idea of HTTP string to check door status?

danb5, 2016-10-12:
Does anyone know what the URL would be for the sensor information?  I am trying to use this with home bridge httpdoor plugin.

jtonzi, 2016-10-12:
Pretty sure you just sub in "upd" instead of "clk".
It'll report back data like this:
{"timestamp": 1476307719, "update": [["left", "closed", 1476305186.770809]]}

danb5, 2016-10-14:
Hi,  I am still looking to get the URL for the status.  The upd mentioned above brings back all doors and all info and I can't filter it by ID.  Is there a way to pull back just the opened or closed status for a select door?

jtonzi, 2016-10-14:
If you specify left or right after the ? doesn't it just give the one door? I'm not sure, I only have one door.

danb5, 2016-10-14:
No it still replies with the information for both doors.   I am entering it as http://ip_addres:8081/upd?id=left
It gives the info for both doors whether I put left, right or nothing after the upd

jtonzi, 2016-10-14:
Good point. Apparently it doesn't matter what comes after upd. It returns the same no matter what. You'll have to wait for input from someone smarter than myself. Sorry

danb5, 2016-10-14:
Thanks for the attempt. I appreciate it.

danb5, 2016-10-15:
Does any one know how to pull a simp,e status reply of open or closed via URL ?

bigwoof, 2016-12-03:
if you are doing this via the command line you can run awk to pull out the status by looking between quotation marks, then you can print the 8th match (in my case) to give a simple output:

`curl -s http://IPADDR:PORT/upd | awk -F'"' '{print $(8)}'`

andrewshilliday, 2017-01-20:
@danb5 it should be relatively easy to add an additional url handler to obtain information specifically on door.  Do you still have interest in the feature?  If so, I  could probably code it up for you.

bigwoof, 2017-01-20:
@andrewshilliday I would be interested in that - the curl/awk is a kind of solution but it would be better to have something a little more robust. 
Thanks for the effort!

andrewshilliday, 2017-01-20:
@bigwoof I just committed the change.  You can now access the state of the door by going to:
http://IPADDR:PORT/st?id=ID

bigwoof, 2017-01-20:
wow that was quick!! 
I will download it, give it a shot, and let you know if I find any bugs.
Thanks a lot!

andrewshilliday, 2017-01-20:
@bigwoof. No prob. I only just realized I've had the wrong email account associated with my github acct all these years which is why I've been so unresponsive.  Now that I actually receive notifications all be able to reply sooner to issues.

danb5, 2017-01-20:
Hi, yes.  I do still have an interest.  Than you so much. I am trying to get the status of the door from a command line in my HomeAssistant (similar to openHAB).   Looking for it to return a 0 or 1 for open or closed. 

Dan

bigwoof, 2017-01-23:
A bit off topic (relating to HomeAssistant):

Hi @danb5 I am also using HomeAssistant to control the garage door.  I still haven't tested it fully but i use the command line to get the state of the door via a template in hass:

```
cover:
- platform: command_line
  covers:
    garage_door:
      command_open: curl -s http://<IP>:8088/clk?id=left
      command_close: curl -s http://<IP>:8088/clk?id=left
      command_stop: curl -s http://<IP>:8088/clk?id=left
# new state command - sitll not 100% tested 
      command_state:  http://<IP>:8088/st?id=left && echo
# old state command that was working 
#      command_state: curl -s http://<IP>:8088/upd | awk -F'"' '{print $(8)}'
      friendly_name: Garage
      value_template: >
        {% if value == 'open' %}
        100
        {% elif value == 'closing' %}
        40
        {% elif value == 'opening' %}
        60
        {% elif value == 'closed' %}
        0
        {% endif %}

```
Sorry not state is not fully tested so it may not work properly.

danb5, 2017-01-25:
Wow, thanks

Dan

closed 2017-07-11 by andrewshilliday