andrewshilliday/garage-door-controller #14: systemd service

flare04, 2015-11-17:
Hi I created a systemd service file for the controller to startup and be monitored by systemd.

# garage - Controls the garage door
#

[Unit]
Description="Garage Door Controller server"
After=network.target auditd.service

[Service]
WorkingDirectory=/usr/local/garage-door-controller
ExecStart=/usr/local/garage-door-controller/controller.py
StandardOutput=syslog
StandardError=syslog
ExecReload=/bin/kill -HUP $MAINPID
KillMode=process
Restart=on-failure

[Install]
WantedBy=multi-user.target
Alias=garage.service

You might want to add it to the repository

For this to work you need to add "!/usr/bin/python" to the top of controller.py

andrewshilliday, 2016-01-03:
This is great.  I'n not very familiar with the systemd setup.  Can you put together some instructions on how to install this and use it?  If so, then I'll be happy to include it in the main trunk.

DervMan, 2017-01-19:
I've got this script working on a Raspberry Pi, here's what I needed.

> root@raspberrypi:/home/pi# lsb_release -a
> No LSB modules are available.
> Distributor ID: Raspbian
> Description:    Raspbian GNU/Linux 8.0 (jessie)
> Release:        8.0
> Codename:       jessie

I added this to the top of controller.py 
`#!/usr/bin/python`

And here's the 'service' file I used (based on the above but needed to add the python exe to the ExecStart line.

`root@raspberrypi:/home/pi# more /lib/systemd/system/garagedoor.service
[Unit]
Description="Garage Door Controller server"
After=network.target auditd.service

[Service]
WorkingDirectory=/home/pi/garage-door-controller
ExecStart=/usr/bin/python /home/pi/garage-door-controller/controller.py
StandardOutput=syslog
StandardError=syslog
ExecReload=/bin/kill -HUP $MAINPID
KillMode=process
Restart=on-failure

[Install]
WantedBy=multi-user.target
Alias=garage.service`

For completeness here's the output from 'Service garagedoor status'

`root@raspberrypi:/home/pi# service garagedoor status
● garagedoor.service - "Garage Door Controller server"
   Loaded: loaded (/lib/systemd/system/garagedoor.service; disabled)
   Active: active (running) since Thu 2017-01-19 11:04:33 UTC; 6min ago
 Main PID: 3425 (python)
   CGroup: /system.slice/garagedoor.service
           └─3425 /usr/bin/python /home/pi/garage-door-controller/controller.py

Jan 19 11:04:33 raspberrypi systemd[1]: Started "Garage Door Controller server".
Jan 19 11:04:37 raspberrypi garage_controller[3425]: we are using SMTP
Jan 19 11:04:37 raspberrypi garage_controller[3425]: Open: unknown => closed
Jan 19 11:04:37 raspberrypi garage_controller[3425]: Close: unknown => closed`

closed 2018-02-15 by andrewshilliday