andrewshilliday/garage-door-controller #15: Emails have no subject

flare04, 2015-11-17:
The emails are sent correctly but do not have a subject.

I'll try and suggest some code soon.

Pack3tL0ss, 2015-11-17:
I added the subject on mine.  My Function now looks like this

```
    def send_opendoor_message(self, opentime):
        syslog.syslog("Sending open door message. (%s)" % opentime)
        self.door_open_text = config["messaging"]["door_open_text"]
        config = self.config['smtp']
        server = smtplib.SMTP(config["smtphost"], config["smtpport"])
        if (config["smtp_tls"] == "True") :
            server.starttls()
        server.login(config["username"], config["password"])
        SUBJECT = "Garage Door Open"
        TEXT = self.door_open_text + " (%s)." % elapsed_time(opentime)
        message = 'Subject: %s\n\n%s' % (SUBJECT, TEXT)
        server.sendmail(config["username"], config["to_email"], message)
        server.close()
        self.msg_sent = True    
```

andrewshilliday, 2016-01-03:
That's a good point.  I've been using the emails to send SMS messages to phones, so a subject was not needed.

jkissner, 2016-10-13:
I added in some lines to over come this issue.  It now sends a full email.  Let me know where I can send the new code if someone is interested.

DervMan, 2017-01-19:
@jkissner Could you upload the code here so I can use it? Thanks

andrewshilliday, 2017-01-20:
@jkissner, if you forked the code, you can issue a merge pull request and I can merge it back in.  otherwise, just send me the relevant code snippet and I can add it to the master.

closed 2017-03-04 by andrewshilliday