andrewshilliday/garage-door-controller #16: Bad wiring schematic

lamping7, 2015-12-22:
The schematic in the readme is potentially dangerous for the RPI. You are mixing 5V logic (required for the relay board) with the 3.3V logic used by the RPI's GPIO. The power to energize the 5V coils of the relay board needs to be isolated from the RPI. Use a separate 5V source, not the one on the RPI. Below is a more correct schematic. I used your image but edited it slightly. NOTE: The jumper bonding VCC to JD-VCC **MUST** be removed.

[image]

The software is great. Thanks.

andrewshilliday, 2016-01-03:
Why would you say that the power to energize the relays needs to be isolated?  The 5v pin is a pass through from the RPI's power source.  Are you afraid of a surge hurting the RPI?  I've had mine hooked up for over two years without a problem.

lamping7, 2016-02-16:
The problem is that with your schematic, you are feeding 5v into the RPI's GPIO. In the Door class (controller.py), the toggle_relay function pulls the corresponding pin to LOW to trigger the relay then sets them back HIGH. This works because the relay is active low. In other words, the RPI GPIO is a sinking digital output. So, when you toggle the relay, setting the pin LOW, current flows from the relay into the RPI. You are sinking 5v into the RPI's GPIO that is built for 3.3v logic. The way I drew it, JD-VCC is used for relay power, VCC is used for logical power to drive the optoisolator, and should be separate. Sure, the RPI might live for a while hooked up the way you have it, but it will most probably lead to damage of the RPI over time. 

You can drive the optoisolator with 3.3v. but not the relays, hence the need for 5v. Some people prefer to swap the operation of the relay, by using a transistor and some resistors to make it active high. You can find more info reading here: www.raspberrypi.org/forums/

LBegnaud, 2016-05-28:
Is there a reason we can't use the 5v output from the gpio instead of an external power source and otherwise follow your updated schematic?

budpi, 2016-06-12:
I have had no issues with the original schematic and setup. I have been running this for about a year and a half on a Raspberry Pi Model B with Motion installed and a Playstation EyeToy for visual confirmation. NO POWER ISSUES. Thanks for this project, it is to-date the most successful project that I have ran on the Pi's.

LBegnaud, 2016-06-14:
@budpi fair enough, but if someone who knows about embedded circuitry says that it's bad to pass 5v through things intended for 3.3v, I'm inclined to believe them.

@lamping7 care to chime in?

lamping7, 2016-07-15:
@LBegnaud You can use the 5v pin, but then you need to consider the total current on circuit. For more info read here: https://www.raspberrypi.org/forums/viewtopic.php?t=14153&p=158827

LBegnaud, 2016-07-15:
@lamping7 I appreciate the detailed response. Didn't think about total consumption on the RPi board...

I'll definitely just go the external power route based on the last link you posted. 

Thanks!

andrewshilliday, 2017-01-20:
@lamping7 Thanks for writing this up and detailing the problem.  The way that I have it set up, is my risk that I'll kill the relay/raspberry pi?  I'm not trying to sound skeptical (in fact you have me quite convinced), but I have been running my garage door set up for a really long time now without a problem.  If I fry an RPI or relay, I'll definitely rewire it the way you suggest.  In the meantime, I'll link to this thread in the main README so that folks can make their own informed decisions.

mew1033, 2017-02-07:
@lamping7 So what would be the easiest way to wire this up correctly? Could I just take a usb cable and cut off the end and connect the 5v and gnd wires to the relay? I'd like to power this with a single outlet (garage ceiling) and I don't know much about circuits...

lamping7, 2017-03-17:
@andrewshilliday The risk is killing the RPi. I don't know what kind of tolerance is built into the RPi, so I can't say definitively if/when damage will occur. I'm just pointing out that there is definitely a risk if you wire it the way you have it, using the RPi's 5v pin and bridging VCC and JD-VCC.

@mew1033 My comment above details how I did it --> https://github.com/andrewshilliday/garage-door-controller/issues/16#issuecomment-232974840

mew1033, 2017-03-17:
@lamping7 So did you just take a single 5v USB power supply, strip the end off and wire it directly into the relay? Then does the relay pass power back to the pi? Or do you power them separately?

lamping7, 2017-03-28:
@mew1033 As I said in that comment: "I used one of the Leviton USB outlets and 2 USB cables to power the system. One goes to microUSB for the Pi and for the other, I cut the end off and soldered the pigtail wires to female pins to plug into the relay board."

mew1033, 2017-03-29:
@lamping7 Ah, I missed it! Thanks a ton for pointing it out. I think I'm going to give it a shot.

monkeyboyohyeah, 2017-05-06:
I have a problem somewhere, somehow. I have wired my pi, relay and reed switch as recommended, but the reed switch 'high' keeps flickering. I'm viewing it in real time with cayenne app.

I've tested 2 Pi's and the reed switch so there's something wrong somewhere else. I've since tried putting a 4.7k resister on the reed signal line to no avail. 

Any help would be appreciated.

monkeyboyohyeah, 2017-05-06:
Actually, just if it matters, I'm using gpio18 and neighbouring earth for input.

gwhiteCL, 2017-06-09:
I'm an EE. Unless you can explain exactly what the risk is to the RPi, I'm going to have to call BS on this.

Per the circuit diagram for the relay board (https://www.sainsmart.com/sainsmart-2-channel-5v-relay-module-for-arduino-raspberry-pi.html) the path between Vcc and the Input Pin is through 3 components:  a resistor (R1), the input side of the 817C optocoupler, and an onboard LED.  The resistor is 1K, and the LED is red. Per the 817C spec sheet, the forward voltage of the 817C is typically 1.2v. A red LED Vf is 1.8v.  So, the total Vf for these two components is ~3v.

If the relay board is powered with 5V as @andrewshilliday originally suggested, then operation will be as follows.  

When the RPi output is high (3.3v), the Vcc-In voltage will be 1.7v (5v - 3.3v) which is less than the total Vf for the 817C & LED, so they will remain off, no current will flow through the optocoupler (or the RPi GPIO pin), and the relay will not be energized.

When the RPi output is pulled low, the Vcc-In voltage will be 5v, which (being greater than 3v) is enough to turn on the 817C & LED.  This leaves ~2v across the R1 resistor (1k), for a current of ~2mA.  This is sufficient to put the optocoupler output into saturation, which results in the relay being energized.  

I confirmed this with my bench supply and multimeter, I'm seeing 1.98 mA (Vcc=5v, In=0v).

Per http://www.mosaic-industries.com/embedded-systems/microcontroller-projects/raspberry-pi/gpio-pin-electrical-specifications , the RPi GPIO pins are capable of sinking 2mA without issue. 

If the RPi GPIO pin is inadvertently put in input mode, I don't see an issue here either.  If the internal RPi (~50k) pull-down resistor is enabled, the RPi input voltage will be ~1.96v, with ~40 uA flowing through the pull-down, no problems there.  If instead the internal RPi pull-up resistor is enabled, the LED and optocoupler diode will prevent current from flowing, so the RPi input voltage will be 3.3v, also totally fine.

So, I'm not seeing any realistic reason that wiring this up as @andrewshilliday originally suggested would cause ANY issue for the RPi.  If you think I've missed something, I'd love to hear it.

The only downside that I can think of for this simple setup is a functional one.  When the RPi boots, my understanding is that the GPIO outputs get pulled low (I haven't confirmed that).  If that is the case, then the relay will get briefly energized whenever the RPi boots, which would activate your garage door opener.

Furthermore, I think connecting Vcc to 3.3v might not result in reliable operation.  With the 3v Vf, that only leaves 0.3v across the R1 resistor, so only 300 uA through the optocoupler.  That may not be enough current to reliably put the output stage into saturation. 

The total current draw from my 5v supply with both relays energized is approx. 120mA.  It seems to me that you can safely drive this off of the 5v GPIO pin as long as the power supply you are using can provide this much current (in addition to whatever the RPi and USB peripherals consume).

andrewshilliday, 2017-06-14:
Thanks for your detailed writeup. My experience is consistent with what your saying.  I've not had any problem with damaged or failing electronics in the several years now that I've been running it. 

I can also say that the boot-time concern does not appear to be an issue.  I reboot my pi all the time and the garage doors do not open whenever that happens.

closed 2017-07-11 by andrewshilliday

pbouthil, 2017-07-30:
FYI, After performing a full install of software and hardware for one garage door (Right) I was having issues with the magnetic contact switch not being recognized on the GPIO21 (Right). After some investigation I found that your schematic is for a GPIO pin out for a RPi Rev.1. If you are using a RPi Rev.2 the GPIO is 27. The config.json will need to be edited to reflect the correct GPIO.

igalesic, 2018-04-29:
Would this work with a Raspberry Pi zero W

savy10, 2018-04-29:
@igalesic yes, I'm currently running it on Raspberry Pi Zero W.

igalesic, 2018-04-29:
Thanks for letting me know. I am going to order one then.

eaglejs, 2021-10-11:
Ah, Sorry, @lamping7 speaks about using an alternative power source for the switch relay, but I haven't seen any good recommendations on what to use instead of a pi. Being that I am not an electrician, I would love to hear of some good recommendations for which power source to use. Kind of how you (@andrewshilliday) had listed where to buy the switch relay and other parts as well. :) Thank you very much!

lamping7, 2021-10-11:
> I did find a link for making a 5v power supply. Would this work?

I'm not sure if it is 100% required, as I didn't do the detailed analysis like in [comment above](https://github.com/andrewshilliday/garage-door-controller/issues/16#issuecomment-307453508) but yes that would work. My first Pi lasted a month, and I guessed this to be the cause.

My Pi is powered via PoE by my switch (which has a UPS) through a PoE splitter to the USB port. 5V then comes from a different 5V USB power supply similar to what's going on in that video. It's been working for years.