andrewshilliday/garage-door-controller #17: sysuptime to add to the Web page

Gilles94500, 2016-01-26:
Hello Andrew,
First of all, thanks for your script. It works fine and is straightforward.
I have an idea to submit :
When the raspberry reboots for any reason, the opening and/or closing date show the same date: the date of the linux boot. 
I have no way to know (remotely) if the doors have been opened/closed or if the linux has rebooted. 
I would like to add the sysuptime (boot date and time) as part of the title of the webpage.
I have tried to update the index.html, but no way. As I am a newbie and as you use many languages (python, html, css, javascript with jquery and Ajax, at least), I am lost and I don't know where to add the code. 
Meanwhile, I have got a lot of training books and videos about these languages but time is missing...

Thanks
Gilles

Gilles94500, 2016-03-18:
Hello,
i have tried to do it and have also asked some friends but we have not found any way to add the sysuptime of the raspberry to the web status page. Is there anybody who could help or give a clue?
Thanks
Gilles

zackoch, 2017-03-24:
The python script uses Twisted to set up a ultra simple web server - it can't render php though. The easiest way to do this would be to use a php shell uptime -p function, but as I stated Twisted can't render php. 

You'd have to re-write the way that the script works with the webserver to use something other than twisted.

zackoch, 2017-03-24:
A quick but really hack-ish way to do it would be to stand up a httpd server on the pi, add php - create a page that uses grabs uptime -p and display it. 

Then create an iframe in a div on index.html with the source of the page you created.

Super jank.

andrewshilliday, 2017-03-27:
There's no need to stand up a http server on the pi.  We're already doing that with twisted python.   So we have two options (neither of which I'm particularly familiar with, but either should work).  For one thing, twisted python can render php (see http://stackoverflow.com/questions/14541813/python-twisted-render-php).   Alternatively, we should be able to add an additional endpoint to serve up the machine start time (or uptime, whichever you prefer) and have a javascript/ajax call retrieve it and render it at the footer.  This would be a bit more complicated, but allow the page to automatically update the uptime and recurring intervals (in case the server is rebooted while the page is open).

andrewshilliday, 2017-03-27:
I just committed a change to the source code to do the second option.  If you update, you should see an uptime status at the bottom of the page.

zackoch, 2017-03-27:
Look at you go. I did see the bit about cgi-php and went.... nooooope - outta my skill level. I had tried native php and found it wouldn't render. I digress.

The solution you came up with is great - I wish I knew JS better. Sorry for the super jank workaround I'd mentioned lol

andrewshilliday, 2017-04-03:
No problem!  Thanks for the suggestions.  Closing this issue.

closed 2017-04-03 by andrewshilliday