andrewshilliday/garage-door-controller #18: [QUESTION] - Possible to integrate SSL?

kyle95wm, 2016-02-07:
Since I plan to open this up onto the internet from my home I was wondering if it would be possible to encrypt the traffic so that one can't sniff the username and password being sent

mjulienabt, 2016-10-18:
I have the same requirement. For now I configured the Apache web server which was already on my pi as an SSL reverse proxy, created self sign cert and key with openssl, enabled the SSL and Proxy modules in Apache, then configure the the ssl and ProxyPass/ ProxyPassReverse directives in the apache.conf file. Only issues was I had to adjust the urls in the index.html file to https to get rig of the mix content warring in the browser.  Seems to work well. The one feature I would like to have is a lockout if too many wrong password attempts are made.

kyle95wm, 2016-10-19:
This is why it'd be useful to have knockd on your garage door server. Only accept incoming connections from your local subnet, and accept incoming connections from devices that used port knocking, while dropping everything else from coming in.

andrewshilliday, 2017-01-20:
This would be a great feature to add.  Unfortunately I have no idea how to do it.  I suspect that the twisted library (which I'm using for the server) supports SSL.  I'll look into it.

dhop90, 2017-05-14:
I've been able to implement https using self-signed certs:  
- Step #1: create self-signed certs
```openssl genrsa > key.pem  
openssl req -new -x509 -key key.pem -out server.pem -days 1000 -subj "/C=<country>
/ST=<state>/L=<city>/O=<org>/OU=<org>/CN=<system name>/emailAddress=<email address>   
```
Step #2: modify controller.py
```
#        reactor.listenTCP(self.config['site']['port'], site)  # @UndefinedVariable

        certData = getModule(__name__).filePath.sibling('server.pem').getContent()
        myContextFactory = ssl.DefaultOpenSSLContextFactory(
               'key.pem', 'server.pem'
               )

        reactor.listenSSL(443,site,myContextFactory)
        reactor.run()  # @UndefinedVariable
```

complete controller.py attached (had to rename file to .txt to upload)
[controller.txt](https://github.com/andrewshilliday/garage-door-controller/files/999470/controller.txt)