andrewshilliday/garage-door-controller #19: How to disable Username/Passoword Authentication

jm19063, 2016-02-09:
Is there a way to comment out the username/password requirement for controlling the garage door?  I'm new at coding, so I'm not sure how.  I'm assuming it's in the following lines:

`clk = ClickHandler(self)
        args={self.config['site']['username']:self.config['site']['password']}
        checker = checkers.InMemoryUsernamePasswordDatabaseDontUse(**args)
        realm = HttpPasswordRealm(clk)
        p = portal.Portal(realm, [checker])
        credentialFactory = BasicCredentialFactory("Garage Door Controller")
        protected_resource = HTTPAuthSessionWrapper(p, [credentialFactory])
        root.putChild('clk', protected_resource)`

Can anyone provide guidance on how to turn off the authentication?

Thank you.

jtonzi, 2016-03-08:
Really late on this reply, sorry.

If you look down the Issues list, you'll see Feedback.  In there it explains some code changes that were made on a different fork and then the config.json changes you need to make.  This way you can turn it on and off with the config file.

andrewshilliday, 2017-01-20:
Almost a year late, but as of the latest source code, you can set use_auth to false in the config.json file and it will not use a username or pw.

closed 2017-01-20 by andrewshilliday