andrewshilliday/garage-door-controller #24: Garage "state" (open or closed icon) not updating

morejava, 2016-05-12:
Thanks for the great code.  This has been fun.  And with Tasker and AutoVoice paired with it, setting up "Ok Google, Open garage door" for voice activation is the cherry on top.  Thanks!

One small issue is that the closed and open icon is not updating.  So I have no idea if the garage is open or closed.  At reboot the icon updates correctly to the current state.  But then after opening or closing the icon remains the same and the timestamp doesn't update.  

Any ideas on what's wrong?

anderson128, 2016-07-28:
I am having a similar issue. The Right and the Left door do not update when I ether open or close the doors. The Right always says open and the Left always says Unknown no matter what pins I setup in the config.json file and what pins I use on my Pi 2. The Relays work on the Pi 2. Does anyone have this working on the Pi 2?

danb5, 2016-10-12:
You need to change the use openhub to false in the config.json
That fixed it for me.

kingie59, 2016-10-13:
Thanks for posting this fix I've changed mine to false and it working now as it should

anderson128, 2016-10-13:
I changed mine to false and restarted but still no luck. No errors in the syslog.

andrewshilliday, 2017-01-20:
Marking as closed.  I've changed the config json to have that turned off by defualt.

closed 2017-01-20 by andrewshilliday