andrewshilliday/garage-door-controller #27: Is this application compatible with Raspberry Pi 3 Model B?

lchacon19, 2016-05-26:


atljam, 2016-05-29:
It works with Pi 3. If you follow the installation instructions and you must edit config.json especially with this new version. change use_openhab to false, if you want to protect the site from anyone being able to open and close your garage door set use_auth to true then change usename and password, next set use_alerts to true if you want notification if you left garage door open,  set alerts = "smtp" I use smtp and use my phone number as the to-address. I created a simple gmail acct as you will be exposing username and password in this file. You will only see the door list if the program is running properly, that's why you see only title., I also use different GPIO pins and only have one garage door so i change that. See my example
[config_json_sample.pdf](https://github.com/andrewshilliday/garage-door-controller/files/288370/config_json_sample.pdf)

flompindonkey, 2016-06-09:
Hi, 

I'm also trying to set this up on a RPi3 and am having some difficulties.  I am a total newbie to everything in this so please bear with me.  So far everything seems to be working except when i pull up the website it doesn't show any buttons. Just "My Garage" banner across the top.  Is My config file in the wrong place? or do i need to point something to it?

Thanks.

atljam, 2016-06-11:
"So far everything seems to be working " why do you think that. navigate to garage-controller-door folder
type command _python controller.py_   and see if there are any errors. Did you review the config.json file I posted above. The website will display even if the python script is not running.

flompindonkey, 2016-06-11:
Thank you for the quick reply.  I guess what i meant was that the apache server seems to be working as i can open the "My Garage" page from outside my home.  When i type command python controller. py it just moves the cursor to the next line and does nothing else.

flompindonkey, 2016-06-11:
pi@raspberrypi:~/¬pi/garage-door-controller $ python controller.py
Traceback (most recent call last):
  File "controller.py", line 332, in <module>
    controller.run()
  File "controller.py", line 212, in run
    reactor.listenTCP(self.config['site']['port'], site)  # @UndefinedVariable
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 495, in listenTCP
    p.startListening()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 979, in startListening
    raise CannotListenError(self.interface, self.port, le)
twisted.internet.error.CannotListenError: Couldn't listen on any:8081: [Errno 98] Address already in use.

So i tried to just redo the whole thing and this is what i got

atljam, 2016-06-12:
well when you typed command and cursor moved to next line that was a good thing, means the program was started, then you should have attempted to use a browser to check your website. in the first case it would appear you had not created a startup entry to get the script running on reboot.
However in the second post seems your error is a result of a conflict of ports so change the port in config.json to some other value like 8085.

closed 2018-02-15 by andrewshilliday