andrewshilliday/garage-door-controller #29: Log file?

theonetmat, 2016-06-21:
Is there a log file or other place where I can see the history of the door actions? 

Thanks for the project. It works great so far.

atljam, 2016-06-24:
cat /var/log/syslog | grep garage_controller
make sure you use the pipe symbol its shift backslash key

theonetmat, 2016-06-24:
Thanks for the idea. I kept looking and did find the syslog and was able to get something similar working. I tried to use a tail-f and then have the grep write to another log file. That does work but only once. What I am doing so far is what I have below. The works but I have to run it manually. I'll eventually have it run maybe once a day to append the new log file. I am not sure what will happen to the file size over time if I keep having it add to the same file.

grep --line-buffered "garage_controller" syslog  >> garagedoorlog.txt

andrewshilliday, 2017-01-20:
Yup.  I put it in syslog out of laziness (and because I really don't know how to do logging correctly in python)  If anyone wants to clean that up they should feel free.

closed 2017-07-11 by andrewshilliday