andrewshilliday/garage-door-controller #30: I tried copying the garage-door-controller in /home/pi/

sandeepbv, 2016-06-23:
When i open the index.html..i am not able to see the anything except "My Garage"..not able to see any icons to select so not sure whether I need to copy the pictures anywhere else along with the folder

Also when i am running the python controller.py..i am getting the below error
pi@sandeep-pi ~/garage-door-controller $ python controller.py
Traceback (most recent call last):
  File "controller.py", line 330, in <module>
    controller = Controller(json.load(config_file))
  File "/usr/lib/python2.7/json/__init__.py", line 290, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting , delimiter: line 28 column 20 (char 459)
pi@sandeep-pi ~/garage-door-controller $

can you please help here ???

sandeepbv, 2016-06-23:
I did not provide any details for the below one 
"openhab" :
{
	"server" : "<SERVER_IP_ADDRESS>",
	"port" : "<OPENHAB_SERVER_PORT>"
}

then I am getting the below 

pi@sandeep-pi ~/garage-door-controller $ python controller.py
Unhandled error in Deferred:
Unhandled Error
Traceback (most recent call last):
  File "controller.py", line 332, in <module>
    controller.run()
  File "controller.py", line 195, in run
    task.LoopingCall(self.status_check).start(0.5)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 173, in start
    self()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 218, in __call__
    d = defer.maybeDeferred(self.f, *self.a, **self.kw)
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 139, in maybeDeferred
    result = f(*args, **kw)
  File "controller.py", line 116, in status_check
    self.update_openhab(door.openhab_name, new_state)
  File "controller.py", line 175, in update_openhab
    conn = httplib.HTTPConnection("%s:%s" % (config['server'], config['port']))
  File "/usr/lib/python2.7/httplib.py", line 712, in __init__
    (self.host, self.port) = self._get_hostport(host, port)
  File "/usr/lib/python2.7/httplib.py", line 754, in _get_hostport
    raise InvalidURL("nonnumeric port: '%s'" % host[i+1:])
httplib.InvalidURL: nonnumeric port: '<OPENHAB_SERVER_PORT>'
Traceback (most recent call last):
  File "controller.py", line 332, in <module>
    controller.run()
  File "controller.py", line 212, in run
    reactor.listenTCP(self.config['site']['port'], site)  # @UndefinedVariable
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 495, in listenTCP
    p.startListening()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 979, in startListening
    raise CannotListenError(self.interface, self.port, le)
twisted.internet.error.CannotListenError: Couldn't listen on any:8002: [Errno 98] Address already in use.

sandeepbv, 2016-06-23:
I tried changing the port to different ones but none of them are working.

my config files are here are 
/etc/webiopi/config
[GPIO]
# Initialize following GPIOs with given function and optional value
# This is used during WebIOPi start process
#21 = IN
#23 = OUT 0
#24 = OUT 0
#26 = OUT 1
7 = OUT 0
8 = OUT 0

[HTTP]
# HTTP Server configuration
enabled = true
port = 8081

/home/pi/garage-door-controller/config.json
"openhab" :
{
        "server" : "<SERVER_IP_ADDRESS>",
        "port" : "<OPENHAB_SERVER_PORT>"
},
"site" :
        {
        "port":7001,
        "username":"xxxxx",
        "password":"xxxxx"
        },
"doors" :
        {
                "left" :
                        {
                        "name" : "LEFT",
                        "relay_pin" : 24,
                        "state_pin" : 17,
                        "approx_time_to_close" : 10,
                        "approx_time_to_open" : 13,
                        "openhab_name" : "g_door_left"
                        },
                "right" :
                        {
                        "name" : "RIGHT",
                        "relay_pin" : 26,
                        "state_pin" : 27,
                        "approx_time_to_close" : 10,
                        "approx_time_to_open" : 13,
                        "openhab_name" : "g_door_right"
                        }
        }
}

loflo, 2016-09-29:
What was the fix for the issue from the first post?

andrewshilliday, 2017-06-02:
My suspicion is that your config file is not valid json.  Can you post the raw config file?

zackoch, 2017-06-30:
Seems to be the result of forgetting quotes in the JSON file.

Poncherelly, 2017-10-07:
I think I;m having the same issue although slightly different (char#).  I simply copied (cloned) from the links provided ... is there a config that NEEDS to be changed specific to each install that we missed?

andrewshilliday, 2017-10-07:
Three is the config.json file that should be changed to your needs.  What is the issue you're having?

closed 2017-11-14 by andrewshilliday