andrewshilliday/garage-door-controller #32: can't send email:  use_smtp always set to false

presck, 2016-07-05:
Hi,  wondering if anyone has come across this problem with sending email.   I installed the controller on my Rasp Pi 3 and everything except email seems to be working.  For some reason, the use_smtp flag is always set to false, and thus prevents email from being sent.  I am somewhat tech savvy, but know very little about python.  I did manage to learn enough pdb to debug a little and found that in the clause that compares set(smtp_params) with set(config['alerts']['smtp']) seems to be the problem.  From what i can tell, the smtp_params variable contains 'time_to_wait', but this is not present in config['alerts']['smtp'] and thus it always sets self.use_smtp = false.  

Any help would be appreciated.

The clause in question is listed below:
if self.alert_type == 'smtp':
            self.use_smtp = False
            smtp_params = ("smtphost", "smtpport", "smtp_tls", "username",
                       "password", "to_email", "time_to_wait")
            self.use_smtp = ('smtp' in config['alerts']) and set(smtp_params) == set(config['alerts']['smtp'])

My config.json file:
{
"config":
{
        "use_auth":true,
        "use_alerts":true,
        "use_openhab":false
},
"alerts":
{
        "time_to_wait" : 10,
        "alert_type" : "smtp",
        "smtp":
                {
                "smtphost" : "smtp.gmail.com",
                "smtpport" : 587,
                "smtp_tls" : "True",
                "username" : "xxxx@xxxxx.xxx",
                "password" : "xxxx",
                "to_email" : "xxxxx@xxxxx.xxx"
                },

anderson128, 2016-08-08:
I am having the same issue an have tried it on the Pi 2 and 3. The sylog says logs that the garage software is using SMTP when you start the Pi but nothing else logs besides the reed switch states and the door button pressing. Any help would be  appreciated.

Thank 

Ken

jkissner, 2016-10-10:
I am seeing the same issue also.  messages shows:
raspberrypi garage_controller: we are using SMTP

but it never emails when garage door is left open.  There are no further messages in the log.  Can anyone give any direction on where to start looking as to why this is happening?

presck, 2016-10-13:
i never figured out why it is happening, but i implemented a workaround.  not sure if it is related, but I've found that i have to reboot the pi ~ every 2 weeks because it stops sending mail.  

changed
   self.use_smtp = ('smtp' in config['alerts']) and set(smtp_params) == set(config['alerts']['smtp'])
to ...
   self.use_smtp = ('smtp' in config['alerts']) and set(smtp_params) != set(config['alerts']['smtp'])

nlthomas, 2016-10-30:
I had the same problem.  I was able to fix this.  Here's how:

On line 96/97 = remove "time_to_wait" as one of the parameters.  It is not a parameter in the smtp function and is not being passed to it, thus the function on the next line was always false. once you remove this, then if the parameters are passed correctly, then you it will work.

nlthomas, 2016-10-30:
I had the same problem. I was able to fix this. Here's how:

On line 96/97 = remove "time_to_wait" as one of the parameters. It is not a parameter in the smtp function and is not being passed to it, thus the function on the next line was always false. once you remove this, then if the parameters are passed correctly, then you it will work.

in sum: 

Change:
smtp_params = ("smtphost", "smtpport", "smtp_tls", "username", "password", "to_email", "time_to_wait")

TO:
smtp_params = ("smtphost", "smtpport", "smtp_tls", "username", "password", "to_email")

andrewshilliday, 2017-01-20:
Good catch.  The time to wait variable was pushed into the common Alert infrastructure.  I'll fix the code.

closed 2017-01-20 by andrewshilliday