andrewshilliday/garage-door-controller #33: openhab setup

cy-kopath, 2016-07-06:
openhab integration seems like a great addition to this project. Which binding in openhab do you use for this integration? I'm having a bit of trouble understanding setting it up on the openhab end. That could be partially because I'm using openhab2.

oenciso, 2016-07-26:
Also need help with the integration. Any help?

DBags77, 2016-10-17:
In the same boat, tried the HTTP binding, MOTT and am.not having any luck.

Pack3tL0ss, 2017-01-19:
Ditto here. @liorbenkiki contributed the code. @liorbenkiki any way you can update the readme to include required configuration for OpenHAB?  OpenHAB has a lot of moving parts, it's not the most intuitive thing to configure.  A brief summary with the related snips from the openhab files related to support would be perfect.

Thanks for your contribution to the project.

liorbenkiki, 2017-01-19:
Thanks for tagging me @Pack3tL0ss. I didn't see this thread until now.  If one of you guys can update the readme I'd appreciate it. If nobody has time to do it I will sometime soon.

There are two approaches to updating garage door state on openhab, you can use either one or both. 
1. Push updates to openhab from this solution. This is the simplest one to get done since the code is all in here. For this you need to specify in the config file the ip address and port to openhab, and make sure you the garage door items in openhab match the names you set in the config file here.  Also make sure password is set to external. I didn't include any authentication in the calls.

2. Pull updates initiated by openhab. This is more complex. The way I tackled this was a js script that is called using http binding that gets the door state and updates the item. Since this isn't part of the solution I push in I won't go into more details right now. 

To send open/close command:
use http binding with the following: >[*:GET:http://{garage_door_pi}:{port}/clk?id=right]

Pack3tL0ss, 2017-01-20:
Thanks, @liorbenkiki I only just installed OpenHab2 and I know very little about it other than what I've hacked together.  right now I have the garagedoor, and openhab gets updates but I haven't figured out how to do the transform/mapping to tell it closed=OFF open=ON... well I've tried but It still considers open/closed invalid values when they come in... or did you configure the item as contact and not switch?  

Not critical either way, but relevant line from the sitemap, items, and if you used transforms... would probably get me there.

Thanks!

andrewshilliday, 2017-01-20:
I've also just recently installed openhab2 and will be integrating my garage doors soon.  When I do I'll edit the readme (if no one else does it first) with instructions.

liorbenkiki, 2017-01-20:
Try this in the sitemap:
Switch item=g_door_left mappings=[open="Open", closed="Close"]
Switch item=g_door_right mappings=[open="Open", closed="Close"]

Pack3tL0ss, 2017-01-21:
This is what I have right now: it seems like I'm close, but it only accepts OFF and ON as valid inputs from the controller.  I haven't figured out the correct way to have openhab accept open/closed, or to have it convert ON=open OFF=closed for the switch item (i've tried a few things, but none are right apparently)

home.sitemap
```
sitemap home label="My home automation" {
    Frame label="Garage" {
        Switch item=Garage icon="garagedoor" mappings=[open="Open", closed="Closed"]
        }     
}    
```
I tried closed="Close" first, but still only accepts ON/OFF

home.items
`switch Garage "Garage" <garagedoor> {http=">[*:GET:http://127.0.0.1:8081/clk?id=left]"} `

I've tried using map files in the transforms folder, but never with any success.  I could adjust controller.py to send ON/OFF but If it's possible on openhab I think it's cleaner there.

liorbenkiki, 2017-01-23:
the item entry should be a String, not a switch.

RolandGPike, 2017-03-19:
Complete beginner here, and I'm trying to operate my garage door from openHAB and see the door's status.  If anybody has managed to get this working can they please post their  .items and .sitemap configs, Also it's not clear how I apply the 'openhab_name' specified in the config.json file to these files.  

This is what I have at the moment:

home,items
[image]

home.sitemap
[image]

I am able to open and close the door with the switch, but the status never shows anything.  Any help is greatly appreciated.

andrewshilliday, 2017-06-02:
I suspect you'd be better off asking that question to the OpenHAB forum.  it's probably something to do with setting up how often OpenHAB should query for status changes.

closed 2017-06-02 by andrewshilliday