andrewshilliday/garage-door-controller #35: Getting a bunch of errors.

DBags77, 2016-08-22:
Hi all.....new guy to the Rasberry pi world.  I have followed these instructions, used all the latest software offered but am having a ton of issues.
This is installed on a fresh install of Rasbian Jesse on a brand new (still has the new pi smell) Rasberry Pi 3.
When i execute the program this is what I get and the relays trip into the closed position.  I have attached the errors I get when I run the program.

Please Help!

Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "copyright", "credits" or "license()" for more information.
>>> ================================ RESTART ================================
>>> 
Unhandled error in Deferred:
Unhandled Error
Traceback (most recent call last):
  File "/home/pi/garage-door-controller/controller.py", line 332, in <module>
    controller.run()
  File "/home/pi/garage-door-controller/controller.py", line 195, in run
    task.LoopingCall(self.status_check).start(0.5)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 173, in start
    self()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 218, in __call__
    d = defer.maybeDeferred(self.f, *self.a, **self.kw)
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 139, in maybeDeferred
    result = f(*args, **kw)
  File "/home/pi/garage-door-controller/controller.py", line 116, in status_check
    self.update_openhab(door.openhab_name, new_state)
  File "/home/pi/garage-door-controller/controller.py", line 175, in update_openhab
    conn = httplib.HTTPConnection("%s:%s" % (config['server'], config['port']))
  File "/usr/lib/python2.7/httplib.py", line 712, in __init__
    (self.host, self.port) = self._get_hostport(host, port)
  File "/usr/lib/python2.7/httplib.py", line 754, in _get_hostport
    raise InvalidURL("nonnumeric port: '%s'" % host[i+1:])
httplib.InvalidURL: nonnumeric port: '<OPENHAB_SERVER_PORT>'

tynick, 2016-08-29:
In which ways did you modify the config file?

DBags77, 2016-08-29:
I just modified the json file as jn the instructions.  I get the same errors with my modified file and the "stock" file.  Im unfortunately  not a great code guy....my takents are in the hardware intergration areas.  K have even tried on a completely fresh install of rasbian with no luck.

kubodhi, 2016-09-13:
Are you using openhab? There's a section in `config.json` where you need to disable it if you're not using it:

```
"config":
{
        "use_auth":true,
        "use_alerts":true,
        "use_openhab":false <---
},
```

Otherwise, if that's set to true, the default settings for the openhab environment in that same config file are nonnumeric (like your posted error reports):

```
"openhab" :
{
        "server" : "<SERVER_IP_ADDRESS>",
        "port" : "<OPENHAB_SERVER_PORT>"
},
```

andrewshilliday, 2017-01-20:
Yup.  My fault.  I shouldn't have left that on by default.  Fixed in trunk now.  Thanks for spotting it.

closed 2017-01-20 by andrewshilliday