andrewshilliday/garage-door-controller #37: Fresh Install - Left Image Broken

zbowman, 2016-09-26:
Moving my garage controller from an older pi to a newer one. In the process, it looks like there's been some code changes since I first installed. Noticing there's username auth, email, and openhab support now. Awesome.

Downside, it looks like the default image for the left garage door isn't working. Only thing I did different from the tutorial is installed to /opt/garage-door-controller vs /~pi/garage-door-controller.

[image]

tynick, 2016-09-26:
Im experiencing the same thing.

danb5, 2016-10-12:
Experiencing same issue and I did not change the install directory.  I have the broken left door icon and the status never updates.  It reads the correct status on boot up but then never updates unless rebooted.

danb5, 2016-10-12:
I figured it out!  Disable open hub in the config.json.  Change it for true to false.

andrewshilliday, 2017-01-20:
Thanks.  It must be throwing an error when if it attempts to connect to an openhab server that doesn't exist.  I'll see if there's an obvious way to get around that.  In the meantime, I've set the config.json to turn off openHab by default

closed 2017-01-20 by andrewshilliday