andrewshilliday/garage-door-controller #44: Loop in controller.py when the "open" mail is sent and the door stays open

Gilles94500, 2017-01-06:
Here are the /var/log/message 
The "Grille" door is opened
Jan  4 14:00:11 raspberrypi garage_controller: Grille: open => closed
Jan  4 17:29:53 raspberrypi garage_controller: Grille: closed => open
5 minutes later a mail is sent (TimeToWait = 900)
Jan  4 17:34:53 raspberrypi garage_controller: Sending email message

The door is not closed, anyway, it shows closed and a closure mail is sent
Jan  4 17:39:23 raspberrypi garage_controller: Grille: open => closed
Jan  4 17:39:23 raspberrypi garage_controller: Sending email message
and alternatively it show open , close, open , close
Jan  4 17:39:24 raspberrypi garage_controller: Grille: closed => open
Jan  4 17:41:58 raspberrypi garage_controller: Grille: open => closed
Jan  4 17:41:58 raspberrypi garage_controller: Grille: closed => open
etc...

Any idea?
Gilles

Gilles94500, 2017-01-08:
The issue doesn't come from the controller.py . It came from the fact that I was using the same gpio number for the door and the light. The light has no status switch.
Bye

andrewshilliday, 2017-01-20:
Ah, that'd do it.  Glad you figured it out.

Gilles94500, 2017-01-20:
To be honest, I thought that it was fixed, but it is not. I still have the same issue. The issue is only with these "wrong" messages. Fortunately, the door is not opening or closing. So it is not urgent. I am still trying to fix the issue. But despite of the trainings that I have attended, it is not easy for me to do the reverse engineering. :)
Gilles

andrewshilliday, 2017-01-20:
So it sounds somewhat as if your contact switches are flakey.  Are they perhaps spaced too far apart or is the wire somehow not securely attached to the pi?

Gilles94500, 2017-01-20:
You are right I also need to check the wires between the door and the pi. I will also add a lot of syslog messages to try to find out where the trigger is. I will do this this weekend.
Thanks

Gilles94500, 2017-01-22:
You were right. I have replaced the contact switch with a new one and now, everything runs fine. Maybe the cause was because it has frozen outside . 
I close the ticket.
Thanks for your script. It works perfectly and, on top of this, it is structured, so it helps to understand .
Regards,
Gilles

closed 2017-01-22 by Gilles94500