andrewshilliday/garage-door-controller #45: CSS : color insteasd of grey ?

Gilles94500, 2017-01-10:
Is there a way to update this CSS grey color so at a glance you know the status of your doors?
Ex : 
RED      when OPEN
AMBER when OPENING
GREEN when CLOSED

I have had a look at  http://demos.jquerymobile.com/1.4.5/icons/index.html but they say that the choice is between black and white.

Thanks
Gilles

Gilles94500, 2017-01-20:
I have not been able to change the css jquerymobile background parameters, so I have used paint instead. I have added a green color to the  closed.png, a red color to the open.png and an amber color to the opening and closing png :)

Gilles

andrewshilliday, 2017-01-20:
Gotcha.  Well I'm glad you figured out a solution. Can you attach the new icons?   I could push them into the master code as an option.

Gilles94500, 2017-01-20:
Here you go.

Gilles94500, 2017-01-20:
[image]
[image]
[image]
[image]

andrewshilliday, 2017-01-20:
Thanks.  I'll see about adding a color option to the code and include these icons.

closed 2017-01-22 by Gilles94500