andrewshilliday/garage-door-controller #47: An open & close button for each door.

DervMan, 2017-01-18:
My garage door opener requires a separate button for open & close. Any chance we can provide the option for this type of door?

andrewshilliday, 2017-01-20:
Can you send a link to the controller?  That might be a bit complicated of an update.  One alternative could be to get a (single button) wireless garage door opener and wire that into the raspberry pi instead of going directly to the opener.

DervMan, 2017-01-20:
Annoyingly my door remote has two buttons too, so although it's a good idea it wouldn't work!

danblade, 2017-02-15:
Andrew has written his code very well. 

I was able to copy and past the lines of code in the config.jason file.

` "left" :
                       {
                       "name" : "CLOSE",
                       "relay_pin" : 23,
                       "state_pin" : 27,
                       "approx_time_to_close" : 15,
                       "approx_time_to_open" : 15
                       },
`

Simply paste it below and change the name "left" to "middle" and change the pin numbers to an unused set. I utilized this excellent code for a commercial door opener which requires stop, open, and close. It works perfectly! (I did go with the sainsmart_4_channel_relay instead of the 2 channel relay.)

[image]

The stop utilizes a break in power so you use the other side of the relay... (if you get into it, you will understand...).

[image]

Andrew, Thank you for writing this!

andrewshilliday, 2017-03-04:
My pleasure!   I'm glad it's working out for you!

zackoch, 2017-03-24:
You could just treat each button like a separate door yeah?

danblade, 2017-03-24:
Exactly what I did.

andrewshilliday, 2017-06-02:
So you're all good then?  OK to close out this issue?

DervMan, 2017-06-02:
I'm making do with a second door icon. It works but I get duplicate emails. Not had time to play. Go ahead and close the ticket as it works for me.

closed 2017-06-03 by andrewshilliday