andrewshilliday/garage-door-controller #48: Error

telss, 2017-02-04:
Hello,
after a few seconds that I open the door contact I get an error and does not update most of the door icon
thanks for help
__________________________________________________________________
Unhandled error in Deferred:
Unhandled Error
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1192, in run
    self.mainLoop()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1201, in mainLoop
    self.runUntilCurrent()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 824, in runUntilCurrent
    call.func(*call.args, **call.kw)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/task.py", line 218, in __call__
    d = defer.maybeDeferred(self.f, *self.a, **self.kw)
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/internet/defer.py", line 139, in maybeDeferred
    result = f(*args, **kw)
  File "controller.py", line 129, in status_check
    self.send_pushbullet(door, title, message)
  File "controller.py", line 178, in send_pushbullet
    door.pb_iden = json.loads(conn.getresponse().read())['iden']
exceptions.KeyError: 'iden'
_____________________________________________________________________-

andrewshilliday, 2017-02-15:
I'll take a look on my end.  I think I'm getting the same error because my notifications are no longer going through to pushbullet.  Maybe their API has changed.

ToddGreenfield, 2017-03-02:
I just set this up on my desk to start testing and am getting the same thing.

andrewshilliday, 2017-03-02:
The problem is that I accidentally checked in the config file with the notification set to "pushbullet". And without a token that won't work.   I'll change the config when I get to a computer but in the meantime set that value to None.

andrewshilliday, 2017-03-02:
The property to be set to none is AlertType

closed 2017-03-04 by andrewshilliday