andrewshilliday/garage-door-controller #52: What did I do wrong?

webby619, 2017-03-30:
Hey so when I do put in (iphost):8080 I get nothing. When I python controller.py in putty (SSH) it bumps down to the next line and hangs there until I closed out of Putty after about 10 min.

What must I be doing wrong? I am using a Pi Zero W

Can't get into :8080 but can SSH in and WinSCP has access.

andrewshilliday, 2017-03-30:
That's strange that you're not getting any errors.  First of all, did you run python as root? (sudo python controller.py)?

webby619, 2017-03-30:
I will try that next BRB

webby619, 2017-03-30:
Ok whatever I have done since the OP it now goes to the my garage page with :8081

webby619, 2017-03-31:
Ok I take that back... what happen was when I input sudo python controller.py in SSH then let that hang there (like the screen grab) It works.
So I have in putty it sitting there, doing it's python thing then go to the browser type in 192.168.11.3:8081 it reads it. Otherwise nothing.

andrewshilliday, 2017-03-31:
I'm not sure I fully understand your last comment?  Are you still having an issue or do you have it worked out?

webby619, 2017-03-31:
When I sudo python controller.py it is working (like the screen grab)
when I close out of Putty/SSH I can't log into 192.168.11.3:8081

So to me it looks like python-twisted is not staying active? Running?

andrewshilliday, 2017-03-31:
That's the expected behavior.  You can't simply run it inside a putty/ssh connection and expect it to live past your closing the connection.   

See the readme for instructions on how to set the controller to automatically start when the RPI is booted.  Copied below:

**Set to launch at startup**
Simply add the following line to your /etc/rc.local file, just above the call to exit 0:
(cd ~pi/garage-door-controller; python controller.py)&

webby619, 2017-03-31:
[image]

Done that unless I have "wrote" it in there wrong. I have rebooted a couple times since too.

andrewshilliday, 2017-03-31:
after your pi boots up, run "ps -ax | python" . What do you get?

webby619, 2017-03-31:
[image]

webby619, 2017-03-31:
Ok to really, truelly I think I have it set up, or at least this problem solved. I had to modify the code for:

(cd ~pi/garage-door-controller; python controller.py)&

to this:
(cd /home/pi/¬pi/garage-door-controller; python controller.py)&

In rc.local

closed 2017-03-31 by webby619