andrewshilliday/garage-door-controller #53: Controller.py not communicating with GPIO pins?

webby619, 2017-04-08:
Hey me again sorry.
So I can't get things working on my Pi Zero W.  I can't get the controller.py working from your scripts to talk to the GPIOs. I have tried a seperate python script that works and "flips the trigger" of the relay.

I have the website working so I know python is up and running with twisted but when I press the garage door button nothing happens. And I think it's all not communicating with the GPIOs
I have searched over and over for what I could be doing wrong. I did not want to make another issue but have been going around in circles now for too long.

-I have re-imaged raspbian onto an SD card, twice
-Python is running at start-up
-I have put in a basic python script that has worked and toggled the relay so I know the pi sees the GPIO's
-Have tried different GPIOs with also changing the json file
-Because it sees the relay (in another python script) I am confident it isn't a soldering issue

andrewshilliday, 2017-06-10:
Hi.  Sorry for the late reply.   Are you certain you are connecting to the right pins?  Would you be able to upload a picture of your connections?   

What version of RPI are you using?  The gpio numbering may have changed depending on your version.

USAFPride, 2017-09-03:
the piZero has a different pinout.  Where GPIO21 is, it is now GPIO27.  Refer to [this image](https://goo.gl/images/wGhFIL).  You can either edit config.json and change to GPIO22 or wire it to the new GPIO21 location

closed 2017-10-07 by andrewshilliday