andrewshilliday/garage-door-controller #55: Only display single door

monkeyboyohyeah, 2017-06-07:
Hi, Firstly, thanks for a great little programme.

I've only got one big garage door and, try as I might, I can't seem to be able to remove the 'right' door from the programme. I see people are able to add doors - any way I can remove a door?

Thanks in advance.

andrewshilliday, 2017-06-07:
Can you upload your config.json file?

monkeyboyohyeah, 2017-06-08:
Almost completely unmodified. I've only changed the name of the left door to "DOOR".

[config.json.zip](https://github.com/andrewshilliday/garage-door-controller/files/1060279/config.json.zip)

andrewshilliday, 2017-06-08:
Ah.  Well then.  In order to remove one of the doors, simply remove the entire second door description block from the config file.

Delete the following:

```
	"right" : 
	{
		"name" : "RIGHT",
		"relay_pin" : 24,
		"state_pin" : 21,
		"state_pin_closed_value" : 0,
		"approx_time_to_close" : 15,
		"approx_time_to_open" : 15,
		"openhab_name" : "g_door_right"
	}
```

Then you should have only a single door button.

monkeyboyohyeah, 2017-06-09:
You'd think that would work!

However, the script fails with the following message;

pi@raspberrypi:~/garage-door-controller $ sudo python controller.py

Traceback (most recent call last):

  File "controller.py", line 386, in <module>

    controller = Controller(json.load(config_file))

  File "/usr/lib/python2.7/json/__init__.py", line 290, in load

    **kw)

  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads

    return _default_decoder.decode(s)

  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode

    obj, end = self.raw_decode(s, idx=_w(s, 0).end())

  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode

    obj, end = self.scan_once(s, idx)

ValueError: Expecting property name: line 56 column 1 (char 870)

andrewshilliday, 2017-06-09:
Please post your resulting config.json.  You've probably got a rogue comma or something.

monkeyboyohyeah, 2017-06-10:
Got it, 
There was a comma that was causing the problem;
Cheers

[image]

closed 2017-06-10 by monkeyboyohyeah