andrewshilliday/garage-door-controller #57: Cron integration

monkeyboyohyeah, 2017-06-26:
How hard would it be to set up a cron job to check if the garage is open at, say, 10pm, and if it's open close it?

Pack3tL0ss, 2017-06-27:
I actually have a heavily modified version of this project (I need to get around to posting my fork, I've had it running for a couple of years).  My door will alert via PushBullet if the door is left open for 10 minutes, It will auto-close after 20 (both configurable), and it will plays an audible alarm 2 minutes before close, then again (different alarm) for the 5 seconds leading up to the auto-close.  A ton of other modifications to meet me need (sends pics when door is open, but only if I'm not home... zoneminder integration, camera view in web interface...) ... should really get it posted.  

A cron job at a specific time should actually be easier, have it run a script to query door status, or if there isn't an easy way to programmatically query the status, Just read the state of the GPIO Reed switch.  Then have it toggle the door if it's open.  May be just as easy to just build a subset of the controller script that reeds the GPIO (get_state function if memory serves) then toggles the door if not closed (toggle_door) vs having a script query then operate the door via the controller.py script and web-interface.  Depends on if there is a way to query status programmatically or not.

andrewshilliday, 2017-07-11:
A cron job would be very easy.  If you want to create a shell script to check the status of your doors and run some action depending on the status, you can obtain the status using curl by browsing to:
http://[URL]/st?id=[ID]
where URL is the hostname of your RPI, and ID is the id of the door you want to check (so just check them each individually)

monkeyboyohyeah, 2017-07-12:
Great news.

So the script would go something along the lines of;

cron -e

0 23 * * 0  user /opt/checkifdooropen.sh

checkifdooropen.sh
curl -i http://ipaddressofgaragedoorcontroller:8081/st?id=1
if <doorstatus> = 'open' then <close door>, else <exit>

(Sorry for my programming – obviously some scripty googling ahead for me!).

monkeyboyohyeah, 2017-08-31:
Finally got this working with the help of a friend who could point me in the right direction.

Here's the final script;
`content=$(curl -u user:pass http://192.168.15.XXXX:8081/st?id=left)

echo “$content”

if [[ $content == *"open"* ]];

then
    curl -u user:pass http://192.168.15.XXXX:8081/clk?id=left
    echo "closing!"
else
    echo “the door is closed and I’m exiting”

fi`

And the cron tab;
`SHELL=/bin/bash
* 22,23,0,1,2 * * * /home/pi/garage-door-controller/doortest`

budpi, 2017-09-01:
I was just going to search how to do this exact thing and then this email notification came in...

However, despite the raspi-cosmological stars aligning, I cannot for the life of me figure this out. Would it be too much to ask for a little bit more elaboration on how to implement this.

Thanks a million!!!

monkeyboyohyeah, 2017-09-01:
Sure.
It took me hours to get it right, and I'm happy to save you the time.

So, one script, called "doortest" (note: no suffix, just the word 'doortest'). I placed this in the garage-door-controller folder. So;

`sudo nano /home/pi/garage-door-controller/doortest`

Paste in;

```
content=$(curl -u user:pass http://192.168.15.XXXX:8081/st?id=left)
echo “$content”
if [[ $content == *"open"* ]];
then
    curl -u user:pass http://192.168.15.XXXX:8081/clk?id=left
    echo "closing!"
else
    echo “the door is closed and I’m exiting”
fi
```

You need to change "user:pass" to the username and password as in config.json, and, obviously, change the XXXX to your internal IP address. Can't remember if I changed the port (8081) from stock, but if I did you'll need to change that to whatever garage-door-controller is running on.

Add a cron job. Firstly I made sure cron was installed as per usual, e.g. apt-get install cron. There were some issues, and I had to --install-missing etc., which didn't seem to work, but I was nonetheless able to enter cron with;

`crontab -e`

The script also needs to be run as Bash instead of Shell so I had to put that in the front. For me I wanted the garage to close at 10,11 12 1 2, every night of the year (overkill... much overkill. I think I'll change to just 9pm :-). So;

```
SHELL=/bin/bash
* 22,23,0,1,2 * * * /home/pi/garage-door-controller/doortest
```
One more thing. The script needs to be made executable. This had me stuck for hours. But it's as simple as;
`
sudo chmod +x /home/pi/garage-door-controller/doortest`

That's it. I tested the script about a hundred times with cron - checking the system time with 
`date`
and setting a cron job to run the next minute. With such a security risk I'd recommend the same approach, although the door should activate only if the url returns 'open'.

Any problems let me know.

budpi, 2017-09-03:
Thanks so much!! I was making a file with a suffix and the cronjob I wrote was not in the format that you have above. This is a great add-on to an already amazing project.

closed 2017-10-07 by andrewshilliday