andrewshilliday/garage-door-controller #58: Door opener wall button not working when RSPi is on.

veloom, 2017-07-02:
Got it all setup and working well with one garage door on a raspberry pie zero w.  Works great except when the raspberry pie is on and controls the garage door, the normal wall garage door opener button won't work.  If I turn off the rspi the button works fine until I turn it back on and it starts the garage opener program.  

Any ideas?

andrewshilliday, 2017-07-02:
Hm.  Could it be that the relay is not correctly opening the circuit after closing it?  Do you have a volt meter (multimeter) to test whether the relay circuit is open or closed?

veloom, 2017-07-02:
Thanks for the quick reply.  Figured it out.  I was having the the issue with the relay being active at startup that SouthernPlayer noted the changes in controller.py for.  Making those changes corrected my issue with the physical button also.

42 gpio.output(self.relay_pin, False) <--was True
72 gpio.output(self.relay_pin, True) <--was False
73 time.sleep(0.2)
74 gpio.output(self.relay_pin, False) <--was True

closed 2017-07-02 by veloom