andrewshilliday/garage-door-controller #59: Email notification if any doors are open at a specific time?

nlthomas, 2017-07-04:
This is not an issue, but a desired feature.  Is there an easy way to get an email at say 9PM if any doors are open so that I know to close them before I go to bed?

monkeyboyohyeah, 2017-07-06:
Closely linked to my query, I'd suggest. These questions both speak to the issue of the door(s) being left open late!

#57

andrewshilliday, 2017-07-11:
It's a good idea and worth looking into.  In the meantime, you could probably write an IFTT rule that runs at 9 and checks the status URL to see whether it returns opened.

zbowman, 2018-08-24:
What's the status url?

zbowman, 2019-01-28:
This is exactly what I needed to find. Is there a means of sending commands into the garage controller while its up and running? If I setup a script to check door status late at night, can I also send the close command through the script without having to use the web interface?

zbowman, 2019-01-28:
Thanks for the feedback. Just having the url's is a big help.

closed 2020-01-28 by andrewshilliday