andrewshilliday/garage-door-controller #62: Wifi Dropping out? Turn Power Management Off - Informational

USAFPride, 2017-09-18:
After you get your device (ZeroW in my case) all nice and bundled up in a box, you may run into a connectivity issue with the wifi disconnecting and not reconnecting necessitating pulling the power. WiFi Power Management needs to be turned off.

Update `/etc/network/interfaces` and the appropriate wlanx device.

`allow-hotplug wlan0
iface wlan0 inet manual
    wireless-power off                                
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf`

After a reboot`iwconfig` should show `Power Management:off`

andrewshilliday, 2017-10-07:
Thanks for the info.  I was also having WiFi connectivity issues in my own setup (because I had a poor signal out in the garage).   Ended up moving the RPI inside and running wires.

USAFPride, 2017-10-08:
I still had some intermittent issues after turning `Power Management` off.  For some reason the pi would still drop signal and refuse to connect.  I added this script `checkwifi.sh` and CRON job to check the status of the WLAN connection and restart it.  This seems to have completely solved the problem. Verify your name/path prior to usage.

`#!/bin/bash

ping -c4 192.168.0.1  > /dev/null

if [ $? != 0 ]
then
    ifdown --force wlan0
    ifup wlan0
fi
`

chmod 755 checkwifi.sh
and `sudo crontab -e `
`*/5 * * * * /usr/bin/sudo -H /home/pi/checkwifi.sh >> /dev/null 2>&1`

Poncherelly, 2017-10-17:
Hello USAFPride, can you share some steps on how to add your script?  I'm not very Linux literate but can follow direction like a champ :)

Thanks in advance.

closed 2018-02-15 by andrewshilliday

ncovercash, 2020-05-07:
Thought I'd note, on Ubuntu Server >18.04 (and likely other distros) that do not use ifupdown (but netplan or something else), the `/etc/network/interfaces` method does not work, you will need to use something like this (or [Poncherelly's solution](https://github.com/andrewshilliday/garage-door-controller/issues/62#issuecomment-335012901), although that's a bit more brute-force) (note, you will need `sudo`):

First, install `wireless-tools` (with apt or whatever).

Then, append to (or create, if it does not exist) `/etc/rc.local` before the `exit` line at the end (if there is one):
```bash
iwconfig wlan0 power off
```

Once you have completed this step, your file should look like:
```bash
# Some stuff here optionally
iwconfig wlan0 power off
exit 0 # optional
```

Finally, make sure the file is executable (`chmod +x /etc/rc.local`).

Now, on each boot, power save will be disabled!  You can confirm this by running `iwconfig`.