andrewshilliday/garage-door-controller #64: Won't open or close when other program is also used

Jswee1, 2017-09-23:
I've got my rpi zero setup with this works fine but when I also use HAP-NodeJS when I open my door using HAP i can't use the web server after when I click it nothing happens rebooting fixes it but it happens again I believe is a gpio issue because it only happens when I've toggled it with HAP once. The other program is linked here https://github.com/legotheboss/YouTube-files/tree/master/Garage%20Solution/Sensor

andrewshilliday, 2017-10-07:
I'm not sure exactly what you're problem is, but if you're running a second web server on the same rpi, then I suppose it could be possible that the servers are contending for the same port.  You can change the port of the garage door controller in the config.py

Jxck-S, 2024-02-21:
I don't know what I was thinking, when I had this issue I was a noob. Crazy to look back in time at what I was doing.

closed 2024-02-21 by Jxck-S