andrewshilliday/garage-door-controller #66: Real Squid here!

Poncherelly, 2017-10-07:
SO I simply do a clone and try to follow all directions but I can't access the webpage.  When I launch a terminal window and manually run cd ~pi/garage-door-controller; python controller.py I get the following:
pi@GaragePi:~ $ cd ~pi/garage-door-controller; python controller.py
Traceback (most recent call last):
  File "controller.py", line 386, in <module>
    controller = Controller(json.load(config_file))
  File "/usr/lib/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting property name: line 55 column 2 (char 878)

Any ideas on what I did wrong?

thanks in advance.

andrewshilliday, 2017-10-07:
I suspect there is an issue with your json config file.  Did you make any changes to it and perhaps forget a double-quote or a curly brace?

Poncherelly, 2017-10-07:
I made a couple of changes but nothing that looks like it was done wrong:

{
"config":
{
	"use_auth":true,
	"use_alerts":true,
	"use_openhab":false
},
"alerts":
{
	"time_to_wait" : 10,
	"alert_type" : null,
        "smtp":
		{
		"smtphost" : "mail.outlook.com",
		"smtpport" : 587,
		"smtp_tls" : "True",
		"username" : "username",
		"password" : "password",
		"to_email" : "email@adress.com"
		},
	"pushbullet":
		{
		"access_token":"<ACCES TOKEN>"
		},
	"pushover":
		{
		"user_key":"<PUSHOVER USER KEY>",
		"api_key":"<PUSHOVER API TOKEN/KEY>"
		}
},
"openhab" :
{
	"server" : "",
	"port" : ""
},
"site" :
	{
	"port":8081,
	"username":"username",
	"password":"password"
	},
"doors" : 
	{
		"left" : 
			{
			"name" : "My Garage",
			"relay_pin" : 23,
			"state_pin" : 17,
			"state_pin_closed_value" : 0,
			"approx_time_to_close" : 15,
			"approx_time_to_open" : 15,
			"openhab_name" : "g_door_left"
			},
		**I did delete the second door ... did this this break something?**
	}
}

Poncherelly, 2017-10-16:
Ok so I started over from scratch and have the website working with both doors still in the code ... will figure out how to hide one of the doors later.

closed 2017-10-17 by Poncherelly

andrewshilliday, 2017-10-17:
Your problem is the extra comma you have towards the end of the config.json file.  the last three braces shouldn't have the comma in there.