andrewshilliday/garage-door-controller #67: door being opened on its own

docsparks, 2017-10-15:
so i've used this the past year, great work.   however over the past few months we've had our door open for us a handful of times.   ive enabled  the password prompt, disabled ssh, use key authentication to make changes to the pi.  i used to see endless attempts via fail2ban, have changed our port for access to some insane number.   i can't figure it out, why this keeps happening.    would the lack of the 10k resistors be any culprit to this? anyone have any suggestions?    this is becoming a huge problem for me.  any direction is appreciated thanks.

clakmurrick, 2018-01-21:
Does yours open on a power cycle? Perhaps your Pi is power-cycling and when it does it activates the relays.

crockernr, 2018-01-30:
I have just had this issue but we just had a powercut. When power was restored the garage door opened..

pyaniz, 2019-04-28:
Don't know if anyone is still looking at this or not but the reason is likely that the relay (in my case at least) is "activated" by default. Hence if you have it connected to Normal Open "NO" it will be force to Closed whenever the pi power-cycles. I inverted the values of the relay in the controller.py and it worked.