andrewshilliday/garage-door-controller #68: [Question] - Possible to Integrate Raspberry Pi Camera Module V2 to Web Page

sqlcursor, 2017-10-18:
Hello,
I have this setup and it's working perfectly.  However, I would like to include the ability to add a frame to the page that would display the current view of an attached Raspberry Pi Camera Module.  The goal would be to avoid closing the door if a pole or some other object is in the way.  Yes, the garage door has a sensor, but sometimes hockey sticks, flag poles, and other objects (the more kids, the more objects) are in the way.  It would be great to have visual confirmation of the area before triggering the motor.

I would love to write the code for this, I just need help understanding into which file it should be added.  I was looking for a straight html file that gets called, but that doesn't seem to be the case.  Somehow it appears that config.json and controller.py are all that drive the local web service.  I'm just unsure of how to add the camera code into those two files.  If anyone could point me in the right direction, that would be great.  When I get it working, I'd love to post it back here to share with this project.

Thanks!

Poncherelly, 2017-10-20:
I second this request.  I've been playing around with the .json and HTML files but only break things and can't get a streaming picture or even a blank picture to display ... any help would help complete this project ...

USAFPride, 2017-10-21:
Your best bet is to get a separate page working correctly showing the camera feed on it's own html page in the www directory. You can access the page via http://gar.age.door.ip:port/yourpage.html.  Once you have that working correctly, then add a menu to the index.html page to point to your camera page (menu controlled by config.json). 
IMO, displaying it on the main controller page would not work due to limited space for people that are controlling 3 or more garage doors.  2 may work depending on the size of the device you are viewing the webpage with.

ronlh1, 2017-10-22:
[index.pdf](https://github.com/andrewshilliday/garage-door-controller/files/1404919/index.pdf)
This is how I did it.
I added a line in the index.html.
 (see index PDF attachment)
[image]

sqlcursor, 2017-10-22:
Thanks for that! What did you use to support the camera/web service? I was worried about conflicting ports, etc...  There seem to be quite a few camera solutions out there and wasn’t sure which one to pick...

ronlh1, 2017-10-22:
I use motion mmal and dydns service.
You can set which port to use in motion