andrewshilliday/garage-door-controller #69: [Question] OpenHab Integration?

Kingabro, 2017-10-29:
Been using this for quite some time now. Thanks. Have started making the move with my devices over to openhab and I just spotted in your config.json that you have a part set out for it. Did you ever get to complete the integration and if so how did it work?

andrewshilliday, 2017-11-14:
It was a while ago that I was working on that and I don't recall that I ever got it working to my satisfaction.  While you experiment with it, please let us know how it goes.

lamping7, 2018-01-03:
I did this a while back. Pretty simple if you just use the HTTP binding and query the status every few seconds. Can also make a call open and close. I didn't try to mess with the stuff in the config he added. I advise not making the twisted server availible from outside the home and use OpenHAB to authenticate your phone (or whatever you use) and interface with the garage door controller web server locally.

Poncherelly, 2018-01-03:
Anyway to get a user guide for dummies to implement this?  Sounds like something I really should have done but only heard about it in this thread.

lamping7, 2018-01-03:
I am by no means an OpenHAB expert, but rather compiled my settings from other examples found online. So, there could be better ways of doing this, but it works, so I never looked into it further.

I have OpenHAB running on the same Pi as this -- hence the loopback address 127.0.0.1. Make sure the port matches your configuration -- 8081 in my case for the twisted server. 

The `1000` in the .items file means that the status is queried every 1000ms. 

Make sure you've enabled the HTTP binding through the PaperUI.

And then, like I said before, turn off port forwarding for the twisted server in your router and follow the guide here -- https://docs.openhab.org/installation/security.html and go here -- https://www.duckdns.org/ for a DDNS service provider if you don't have one. Don't worry about signing your certificates unless you're buying a domain. There is a setting in the OpenHAB app (at least on Android) to ignore the warnings of a self signed cert.

Hopefully this helps and gives you a good starting point.

.sitemap:
```
Frame label="Garage"
{
	Switch item=GarageDoorButtonLeft mappings=[ON="Toggle"]
	Text item=GarageStatusLeftDisp		
	Switch item=GarageDoorButtonRight mappings=[ON="Toggle"]
	Text item=GarageStatusRightDisp
}
```
garage.items:
```
String GarageDoorButtonLeft "Left Door" <garage_detached> {autoupdate="false"}
String GarageStatusLeft "[%s]" {http="<[http://127.0.0.1:8081/st?id=left:1000:REGEX((.*))]"}
String GarageStatusLeftDisp "Left Door Status [%s]" <garagedoor>
String GarageDoorButtonRight "Right Door" <garage_detached> {autoupdate="false"}
String GarageStatusRight "[%s]" {http="<[http://127.0.0.1:8081/st?id=right:1000:REGEX((.*))]"}
String GarageStatusRightDisp "Right Door Status [%s]" <garagedoor>
```
garage.rules:
```
rule toggleGarageDoorLeft
when 
	Item GarageDoorButtonLeft received command
then  
	sendHttpGetRequest("http://127.0.0.1:8081/clk?id=left")
end

rule toggleGarageDoorRight
when
	Item GarageDoorButtonRight received command
then
	sendHttpGetRequest("http://127.0.0.1:8081/clk?id=right")
end

rule garageStatusLeft
when
	Item GarageStatusLeft received update
then
	if (GarageStatusLeftDisp.state.toString.toUpperCase != GarageStatusLeft.state.toString.toUpperCase) {
		GarageStatusLeftDisp.postUpdate(GarageStatusLeft.state.toString.toUpperCase);
	}
end

rule garageStatusRight
when 
	Item GarageStatusRight received update
then
	if (GarageStatusRightDisp.state.toString.toUpperCase != GarageStatusRight.state.toString.toUpperCase) {
		GarageStatusRightDisp.postUpdate(GarageStatusRight.state.toString.toUpperCase);
	}
end
```

Poncherelly, 2018-01-04:
When trying to install Open HAb I get the following:

@SmartGarage:~ $ wget -qO - 'https://bintray.com/user/downloadSubjectPublicKey?username=openhab' | sudo apt-key add - sudo apt-get install apt-transport-https
gpg: can't open 'sudo': No such file or directory
gpg: can't open 'apt-get': No such file or directory
gpg: can't open 'install': No such file or directory
gpg: can't open 'apt-transport-https': No such file or directory

The command was copied directly from the OpenHAB website so I'm not sure what I did wrong.

Gilles94500, 2018-04-25:
I have tried to use the sample above with openhab2:
2 things: 
**xxx.Sitemap**: the fist statement must start with  the keyword "sitemap' ex : sitemap Garage label="My Garage" 

There is a small bug in the openhab part of the controller.py:
    def update_openhab(self, item, state):
    ......
+        conn.request("PUT", "/rest/items/%s/%s" % (item, state))
-        conn.request("PUT", "/rest/items/%s/state" % item, state)

I am totally new with openhab so I try to adapt the given code. 
Thanks to Lamping7

lamping7, 2018-04-25:
@Gilles94500 
1. Above is only a snippet from my complete .sitemap file, which has many other things, like a frame for my thermostat controls.
2. Because I'm using the HTTP binding to query the garage door controller twisted server, I'm not using the openHAB portion of `controller.py`.

Gilles94500, 2018-04-26:
@Lamping7 
Thanks, so I could disable the openhab section in the config.json file.
I know that it is not an openhab forum, but I don't know how to ask this question in the openhab forum because I have not added this using the openhab menus. Based on your extract, how can I add a color (green when closed and red when open), because everything is grey. (Or point me to a link)
[image].

Thanks
Gilles

lamping7, 2018-04-26:
What do you mean? You want the icons to change colors?

Gilles94500, 2018-04-26:
Yes , I would like to use different icons based on the status.

Until now, I used the controller.py script developed by Andrew. As I am getting old I use a set of icons (green icon when the status is “closed”, amber when “opening” or “closing” and red when “open”) . I prefer to see a color that to try to read a status:) .  

For security purpose, as you have explained above, I would prefer to connect to  Openhab instead. On top of that, I think that Openhab is able to show the status of my SONOFF devices.

So I have set up a Ubuntu and have installed Openhab. I am trying to make it work. Later, I will install Openhab on the raspberry.

Gilles

lamping7, 2018-04-26:
I did this for one of my thermostat status items. Basically you need to create your own icons using your favorite image editor -- gimp for Linux works well. Then you have to name them appropriately... It took me a little while to figure this out. If your going to make your own icons, I can share a working example of how to use it.

Gilles94500, 2018-04-26:
I have made 4 colored icons  that I use with the controller.py (garage-door-controller\www\img\). 

I have read in the openhab doc that to use my icons, I need to copy them in the  $OPENHAB_CONF/icons/classic/ directory but  I don’t know how to set them depending on the status(closed, opening, open, closing, closed). If you could give me some hints it would be very helpful.

lamping7, 2018-05-02:
So the magic happens in .rules file where it sets the string that is used to find the icon file defined in the .items file.

```
rule "EquipmentStatus"
when
  Item ecobee_equipmentStatus changed
then
  if ((ecobee_equipmentStatus.state != null) && (ecobee_equipmentStatus.state != NULL) 
        && (ecobee_equipmentStatus.state != "")) {
    switch ecobee_equipmentStatus.state.toString {
      case "fan"		: ecobee_runningEquip.postUpdate("fan")
      case "compCool1,fan"	: ecobee_runningEquip.postUpdate("cool")
      case "auxHeat1,fan"	: ecobee_runningEquip.postUpdate("heat")
    }
  } else {
   ecobee_runningEquip. postUpdate("off")
  }
end
```
Inside my .items file is (the <> is the important part):
```
String ecobee_runningEquip		"Status" <hvac>				{ autoupdate="false" }
```
Icons are here and named to match the rules:
[image]

Gilles94500, 2018-05-02:
Got it.
I have renamed my icons accordingly and it, woks.

```
Y:\>cd icons

Y:\icons>dir
 Le volume dans le lecteur Y s'appelle openHAB2-conf
 Le numéro de série du volume est 4910-FB5D

 Répertoire de Y:\icons

24/04/2018  10:29    <REP>          .
30/04/2018  09:03    <REP>          ..
28/04/2018  17:14    <REP>          classic
               0 fichier(s)                0 octets
               3 Rép(s)  76 237 279 232 octets libres

Y:\icons>cd classic

Y:\icons\classic>dir
 Le volume dans le lecteur Y s'appelle openHAB2-conf
 Le numéro de série du volume est 4910-FB5D

 Répertoire de Y:\icons\classic

28/04/2018  17:14    <REP>          .
24/04/2018  10:29    <REP>          ..
14/01/2017  08:17               859 garage.png
18/12/2017  14:44               284 readme.txt
14/01/2017  08:17             1 042 garage-opening.png
14/01/2017  08:16               955 garage-closing.png
14/01/2017  08:16               890 garage-closed.png
14/01/2017  08:17               859 garage-open.png
               6 fichier(s)            4 889 octets
               2 Rép(s)  76 237 279 232 octets libres

Y:\icons\classic>
```