andrewshilliday/garage-door-controller #71: Hardware explanation

grosch, 2017-11-28:
Just a suggestion.  Most people looking at this, like me, probably have zero experience with hardware type wiring like this.  I started at your diagram for quite a while trying to understand what was going on.  

I was trying to figure out how the relay was passing a command all the way through to the door opener.  I know, you're laughing now :)  But maybe just add a little text like:

The relay goes to the left side of the GPIO and the software then watches those two GPIO ports to determine whether the door is open or closed.  The right side of the GPIO are the output, unrelated to the input, where the software will send a command to the garage door openers to open or close.

Also, I notice there is a big pull request that looks like it wasn't merged.  Are you going to include that?

andrewshilliday, 2018-02-15:
Hi. Sorry for the delayed response.  

I do need to look at the wiring diagram and make changes.  Especially since it was written for older versions of Pis (I'm not sure what differences there are in new pis, so if anyone has some feedback on what needs to change that would be appriciated).  As to the merge request, the branch was a very useful change, with better organization of the code.  However it also introduced additional dependencies and changed the required version of python.  I felt it was to invasive of a change, especially to folks who already have it working.  Currently no plans to merge it into master.

closed 2018-02-15 by andrewshilliday

ncovercash, 2020-05-31:
Hey – thought I'd pitch in here since I encountered confusion with my Pi v3.  Most new Pis have a 40-pin header which has a bit of a different format, so I would recommend an additional diagram for those!  Also, GPIO 27 is mislabeled as GPIO 21 currently