andrewshilliday/garage-door-controller #72: Website no longer responding

Poncherelly, 2017-11-28:
So I've been up and running for just over two weeks now and it's been running flawlessly.  I woke up this morning however to the website no longer loading.  My first thought was that the server crashed or the wifi failed but I can log on to the server to view the camera is still running.  

Any idea what might have failed or how I can get on these system to find the error to fix it so I don't have to deal with this every two weeks?

If there are no ideas, I'll reboot the raspberry pi to get up and running again and hope that it was a one time glitch.

Poncherelly, 2017-11-29:
So when I got home I rebooted the Pi (unplugged and reconnected) and my website came back up.  I guess now I'll wait another >2 weeks but <3 weeks to see if the problem comes back or if it was a glitch.  fingers crossed this isn't an ongoing problem ...  if it is, I might have to add a WeMo plug in line so that when this happens I can remotely power it off/on so that I have access when needed.

Scenario happened yesterday.  I took my car in to the shop and they kept it overnight.  I don't have an external keypad on my garage door and I use my car's built in garage door opener to get in.  Due to the website going offline, I wasn't able to let myself in and had to wait for my wife.

If this issue resurfaces, can someone write a cron job (similar to the wifi up script from another thread) that looks for it's own webpage and if the website isn't available, it emails me and reboots the system?  Probably best to use a 30 min wait time ...

Poncherelly, 2017-12-01:
So a repeat incident happened but this time around 24 hours after the last reboot.  I rebooted once again and now I'm waiting to see if it becomes unresponsive again.  Similar to the fist time, although the website is unavailable, the camera is still accessible so it's not the WiFi or the raspberry Pi that's down.  seems to just the the web service for the main website that controls the door.

Any thoughts ...

amccombsahce, 2017-12-01:
I haven't gotten a chance to install this project yet because I have the same issues with the weather station setup. I was suspecting either Raspberry Pi 3, Linux Noobs or Apache is garbage.  
The weather station runs php scripts, which I have found that 100s of these scripts can spin doing nothing. I have modified the php scripts to be smarter than what came out of the box,  I also don't leave the browser window open 24/7 to show me the weather update every minute, as well as I do check for software updates more often now,; which seems to help a little to prevent it from becoming non-responsie ever-other-week..

andrewshilliday, 2017-12-01:
Hi. I have had similar issues in the past. It turned out to be very poor Wi-Fi connectivity out to the raspberry pi. I ended up moving the raspberry pie into the house and running some wires for the sensors and relays. Apparently the raspberry pi does not do a very good job with reestablishing connectivity once the Wi-Fi signal is lost.  Is there a garage pretty far from your Wi-Fi hub?

Poncherelly, 2017-12-01:
Although I would lean to think the same way I can say that the WiFi is connected since I'm able to view the pi camera without issue when the site is down.

For example, the garage website accessed via http://url:port1 and the camera can be accessed via http://url:port2.  When I get 404 trying to access http://url:port1, http://url:port2 is still working.  

This tells me that WiFi is live and it's just http://url:port1 that's disconnected for some reason.

amccombsahce, 2017-12-01:
I use hard-wired connections only, good question.

andrewshilliday, 2017-12-01:
So one thing you might want to try is to run the controller.py file manually instead of as a service. That way you can capture the output and see if it’s throwing a message. From the terminal (I recommend doing it inside a screen) kill the running python process and then run sudo python controller.py from the controller directory.

andrewshilliday, 2017-12-01:
You might want to send the output to a file so it’s all saved off.

Poncherelly, 2017-12-20:
Thanks Andrew ... I have the box turned off for now while I print a new case for it.  I also built a dual garage one for my brother in law and that one also experienced the website non responsive but I have access over SSH and the camera stream is still live as describe above.

I'm not sure if its come across this way or not but I'm not a Linux guy at all and rely heavily on google and this forum.  If you could provide step instructions that I can run over SSH, I'm happy to follow and provide any information you would need to help :)  Also thanks again for everything.

Poncherelly, 2017-12-21:
I've been googling for an answer and still can't find one.  I ran in to some instances of Apache needing to be restarted but after trying to find the version of Apache on the system, I'm now assuming that Apache isn't being used??
Everything still works with this box except for the page not being available.  Even the streaming camera is working and I'm SSH'd in to the box and running commands without issue.

Can someone help me run (commands)
1) kill the running python process 
2) run sudo python controller.py 
and how to I send data to an output file to share?

andrewshilliday, 2017-12-22:
1) no, the garage controller does not use Apache.  It uses Twisted Python libraries. 
2) to kill the python process, run "ps -ax | grep controller".  That should give you the id number (PID) of the process.  IT should look something like python controller.py with a number in front. To kill it, run "kill [PID]" where [PID] is the number from "ps -ax | grep controller"
3) when running the program manually from the command line, you can (by running "sudo python controller.py", you can save the output to a file by doing the following:
"sudo python controller.py > outputfile.txt"
Or, if you want to all see it, you can run
"sudo python controller.py | tee outputfile.txt"
that will print the output but also save it.

Poncherelly, 2018-01-02:
So this worked ... I was able to restart the website without having to reboot ... thank you.

I did try and get the output but the file was blank and I was able to get a screen capture but it would appear I rebooted and didn't have the file :(.  Next time the page fails, I'll retry and provide any output information I can pull.

closed 2020-06-18 by andrewshilliday