andrewshilliday/garage-door-controller #79: Door States will not show anything but "closed"

clakmurrick, 2018-01-21:
I'm not sure where to begin with the diagnosis, but the garage doors open and close with the app, however, the states do not change. It was working and one day ceased to work. The wiring has not been tampered with, nor the magnetic switches. I've attempted using different state-pins and editing the config.json accordingly.

I have a Raspberry Pi 3 Model B. Not sure if this is related, but when the Pi is powered on, the red LED remains lit up, but every few seconds or so at random it blinks off then right back on. When I check the website, it still remains up and isn't shutting off, so I'm not sure if I have any power issues.

Feedback would be appreciated! Thank you so much.

clakmurrick, 2018-01-21:
Just attempted using the default state-pins in the tutorial, and I directly connected the state-pin to a ground on the Pi directly with a jumper and the state still wouldn't change. I tried a different power source and still no bacon.

tubastu, 2018-02-12:
If you have configured alerts to use smtp on operation try setting the alerts back to NULL. I never could get my status to update if the alert send didn't complete, it stalls until next button press. I'm still looking for some better python code to handle the messaging. Hoping someone will post something to replace the existing code.

clakmurrick, 2018-02-12:
Are you referring to `"alert_type" : null,` as mine is set to null.

andrewshilliday, 2018-02-15:
I think this is all due to poor error handling of the notifications code.  I've updated the code, but haven't had the opportunity to test it.  Please let me know if this does the trick.

Sparkster971, 2018-08-29:
I'm having the same issue as clakmurrick. I'm setting up a new controller for my son's garage and everything is working great except the state change. I also have it set as "alert_type" : null. Also, use_alerts is set to false. I've tried two different magnetic switches from the pack of a dozen I bought and no joy.

wassimf87, 2019-06-16:
I have the same problem. icons and text don't update/broken. 
Garage state is stuck at what ever the initial status is after reboot.

wassimf87, 2019-06-16:
> I have the same problem. icons and text don't update/broken.
> Garage state is stuck at what ever the initial status is after reboot.

Woke up the next day and it's magically all working properly now. 
FYI, alert type is smtp and modified the code to limit alerts only between 10pm and 5am.