andrewshilliday/garage-door-controller #85: Hide page until logged in

ej21, 2018-04-09:
Hi all,   Anyway the homepage and current status of garage doors could be hidden until username and password provided?

andrewshilliday, 2018-04-09:
It's certainly possible.  I'm not sure it would be preferable for the majority of users, but it could be added as an option.

ej21, 2018-04-09:
Option would be great.  Thank you

jdolence999, 2019-10-22:
I am late to the party putting this project to use since it looks like Andrew wrote the code over 6 years ago.  I got it working with a Camera stream, but I do not want to expose anything to the Internet with out authenticating the user before the home page is loaded, too.  

I am not a python programmer, but I have spent several weeks trying to figure out how to password protect the homepage with authentication.  It would be nice to ensure both the homepage and clicking on the garage door would require authentication.  I'm sure my wife is getting fed up.

I am hoping Andrew still monitors this site.  I am willing to do the work, if someone can shed some light on how to do it.  I have looked at every twisted authorization site and I can get the 'clk" to authorize or prior to the main site (breaks the site), but not just the main site or both.

ijaron, 2020-06-12:
Just leaving this here for anyone else that might have the same issue that they wish to protect the entire page.
I'm using [vouch-proxy](https://github.com/vouch/vouch-proxy) via nginx reverse proxy to secure the controller. Mostly just using the sample setup from the projects readme. It doesn't give you a login for itself but rather allows you to authenticate against another source (e.i. you can use your google login and then set your email as the only allowed login)
Going that route also allows me to have the reverse proxy handle 100% of the ssl and out of network traffic.

jdolence999, 2020-06-16:
That looks like a viable solution.  I fixed the python script to authenticate before showing anything and incorporated a camera stream into the solution.  The downside is that the authentication does not use another source or 2-factor.