andrewshilliday/garage-door-controller #93: Browser issue since the migration to HTTPS

Gilles94500, 2018-06-25:
I have just migrated to https/ssl. It works fine from an Android browser, but the doorlist along with the  status is not shown in Chrome or IE from a Windows PC.

It seems that, as the "index.html" refers to several HTTP links, they are considered as "not secure" .
```
<link rel="stylesheet" href="http://code.jquery.com/mobile/1.2.1/jquery.mobile-1.2.1.min.css" />
    <script src="http://code.jquery.com/jquery-1.8.3.min.js"></script>
    <script src="http://code.jquery.com/mobile/1.2.1/jquery.mobile-1.2.1.min.js"></script>
```

Michael-W, 2018-06-25:
If you make the links "https" it should work.  The files are on the pi in the directories under www.  An earlier issue or commit mentioned making the references local so that operating the doors is possible without an internet connection.  If you change the links to "`www/js/jquery-1.8.3.min.js`" for example it may work too.  I have not done it **yet**!

Michael-W, 2018-06-25:
`  <script src="/js/jquery-1.8.3.min.js"></script>
  <script src="/js/jquery.mobile-1.2.1.min.js"></script>
  <script src="/js/date.format.js"></script>
  <script src="/js/client.js"></script>
`I don't have to try it.  This is in my index.html.

Gilles94500, 2018-06-26:
Thanks Michael,

I will update the www (sub)directories then.

I had not been noticed about the commit. I need to check my github settings :).

 

Gilles