andrewshilliday/garage-door-controller #95: No json object could be decoded error

commputethis, 2018-08-14:
I did a fresh install of Raspbian and followed your directions, but the website didn’t come up after setting it to start automatically and rebooting.  So I ran “python controller.py” from the garage-door-controller directory and get a “no json object could be decoded” error.  I tried redoing everything and tested prior to modifying the config file and still got the error.  I wanted to make sure my edits hadn’t broken anything.  Thoughts?

[image]

commputethis, 2018-08-15:
I am not sure exactly what the issue was, but I took the contents of config.json and ran it through the json formatter/validator at https://jsonformatter.curiousconcept.com/ and then updated the file with the output on that site and now it is working, at least without SSL.  It isn't working for SSL still, but looking into that issue.

commputethis, 2018-08-15:
I found the issue.  Using the Files Compare Tool and comparing the old file to the new file I found there are 3 characters at the beginning of the file that were causing a problem: ï»¿

Removing the characters resolved the issue.  I couldn't see those on the pi itself or in NotePad++.

[image]

andrewshilliday, 2018-08-15:
Glad you got it working.

closed 2018-08-15 by andrewshilliday