andrewshilliday/garage-door-controller #98: Can't send email

hamanjam, 2018-11-14:
First off, thanks for maybe the simplest setup I've seen for this.  I'm getting an error when trying to send email notifications.    Whn I start up, I can connect and everything looks great.  When I disconnect the sensor to simulate an open door, I get a message on the command line after several minutes:

root@ol7:/home/pi/garage-door-controller# /etc/rc.local 
My IP address is X.X.X.X
root@ol7:/home/pi/garage-door-controller# 
root@ol7:/home/pi/garage-door-controller# Unhandled error in Deferred:

Here is my config.json with the email settings:

{
    "config":{
        "use_https":true,
        "use_auth":true,
        "use_alerts":true,
        "use_openhab":false
    },
    "alerts":{
        "time_to_wait":10,
        "alert_type":"smtp",
        "smtp":{
            "smtphost":"smtp.gmail.com",
            "smtpport":587,
            "smtp_tls":"True",
            "username":"fromuser@gmail.com",
            "password":"mypass",
            "to_email":"touser@hotmail.com",
            "subject":"Garage door test"
        },

amccombsahce, 2018-11-14:
Have you tried smtpport 465?

hamanjam, 2018-11-20:
Started completely over and got the same error. Searching turns out that's a problem with a version of "twisted"  I was able to run "pip install Twisted" and cleanup with more info.  Now, the app hangs and stops responding to door events but at least I get more in /var/log/messages:

Unhandled error in Deferred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/base.py", line 1267, in run
    self.mainLoop()
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/base.py", line 1276, in mainLoop
    self.runUntilCurrent()
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/base.py", line 902, in runUntilCurrent
    call.func(*call.args, **call.kw)
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/task.py", line 239, in __call__
    d = defer.maybeDeferred(self.f, *self.a, **self.kw)
--- <exception caught here> ---
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 151, in maybeDeferred
    result = f(*args, **kw)
  File "controller.py", line 136, in status_check
    self.send_email(title, message)
  File "controller.py", line 178, in send_email
    sys.syslog("Error sending email: " + str(inst))
exceptions.NameError: global name 'sys' is not defined

hamanjam, 2018-11-21:
Found the first issue I'm having:

Line 178 in controller.py:
            sys.syslog("Error sending email: " + str(inst))

Change to:
            syslog.syslog("Error sending email: " + str(inst))

Now, in var/log/messages I see the following but no crash (no email but making progress)

Nov 21 17:43:00 garagedoor garage_controller: LEFT: closed => open
Nov 21 17:43:10 garagedoor garage_controller: Sending email message
Nov 21 17:43:10 garagedoor garage_controller: Error sending email: getaddrinfo() argument 2 must be integer or string

hamanjam, 2018-11-22:
Fixed.  My problem in config.json was the port number was in quotes from testing previously.  Once I set up the app password in gmail, I was able to send email.

closed 2020-01-28 by andrewshilliday