andrewshilliday/garage-door-controller #99: Unable to get IOS App working.

DervMan, 2018-12-24:
I've been using this great code for a while now to control my garage door via a website. I see there's now an IOS app for controlling the door, which I've installed and configured my door. However despite having the correct settings in the IOS app all I get is a white screen, I'm not seeing my door listed. 

While troubleshooting I ran a packet capture on the work firewall and the IOS app isn't even trying to access the garage door page, I am using TCP port 888 rather than the default 8081 so maybe the app doesn't like that?

chrisbergeron, 2018-12-27:
I'm having the same issue.  I just paid for the app.  I can load the webapp fine in Safari on the iphone.  I can even hit the endpoints (/upt, /clk?id=right, etc) and toggle the python app.  I've tried with and without SSL, and with/without user credentials; I updated the config.json and the settings in the iphone app and restarted.

I'm on iOS: 12.1 (16B92) - iPhone 7 Plus
My Garage App Version: 1.0.0 (2)

I really dig the codebase and it works great.  I just wish I could get the iOS app to work.  It just displays "My Garage" at the top and the settings icon in the top left corner.  At the bottom, I can see the CSS background for the uptime bar, but no text.

I bought the app to support this project and what you've done so far.

andrewshilliday, 2018-12-27:
Hi all.  I’m sorry you are having trouble with the app, especiallly since you paid for it.  I probably should have been more clear on the main page, but the iOS app is not actually affiliated with this project.  It was developed by another user. My only interaction with the developer was to express my thanks for building it and to ask him to consider giving it away for free instead of charging for it (he replied that he has his Apple developer license to pay for so can’t give it away).  

Hopefully he will come across this messag and fix the issue. If not I can try to dig up his email and reach out to him.

DervMan, 2018-12-27:
Ah I see, that would explain it! Let's hope he can throw some light on this.

USAFPride, 2019-01-27:
You really don't need the app.  iOs instructions:
1. Browse to the homepage in Safari using `http://yourdoorcontrollerip:888` (hopefully you set it up as a static IP)
2. click the box with the up arrow at the center bottom of screen and pick "Add to Home Screen" and follow the prompts

mandogon, 2019-03-02:
USAFPride,
was thinking about getting the app but thanks for the tip added it my home screen and now it looks like the app itself.

zbowman, 2019-03-23:
> I'm having the same issue. I just paid for the app. I can load the webapp fine in Safari on the iphone. I can even hit the endpoints (/upt, /cfg, etc)

What are all the endpoints?

chrisbergeron, 2019-03-23:
/upt - uptime
/upd - update display
/cfg - status
/clk?id=left - door1
/clk?id=right - door2
/st?id=left - view status door1
/st?id=right - view status door2

I installed an SSL cert on my webserver and it started working.  I haven’t had a problem since.

closed 2020-01-28 by andrewshilliday