andrewshilliday/garage-door-controller #100: Can't send email

rothda2, 2018-12-31:
Thanks for sharing this!  I'm waiting for my relay board and then can test, but so far everything seems to be working other than the email settings.  Every time I enter smtp as the alert type in config.json ("alert_type":smtp,) and restart the program it fails with some errors.  Can anyone help please?

pi@raspberrypi:~ $ cd ~pi/garage-door-controller; python controller.py
Traceback (most recent call last):
  File "controller.py", line 425, in <module>
    controller = Controller(json.load(config_file))
  File "/usr/lib/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

rothda2, 2019-01-02:
So, new update - I put smtp in quotes and it gets a little further, but still kicks out an error - "error in deferred".  Any thoughts?

andrewshilliday, 2019-01-02:
Hi.  Can you post the full error?

rothda2, 2019-01-03:
Actually things appear to be working now.  I ran the code through a 2 to 3 converter and that seems to have fixed my issue.   After the conversion I did have to go back and make some minor tweaks like changing http.client back to http.lib, but at least it works now!

closed 2019-01-22 by andrewshilliday