andrewshilliday/garage-door-controller #101: Log tab on the website

mdbrewer, 2019-01-03:
Feature request:
Have a tab on the website that displays the log info either from syslog (grep garage_controller) or the files dumped to a specific file.

NCAvian, 2019-02-16:
This assumes the user that is running the controller is the user "pi".  If you've set up another user on your Raspberry Pi, you'll need to change the user accordingly.  Where the notes say "pi" below, substitute the user you've configured.

As the user pi, do the following:

`cd /home/pi`
`touch garage-door-controller/www/garage_door.html`

Create the file 'command' in /home/pi with your text editor.  It will contain the following:

`#!/bin/bash`
` `
`cat /var/log/syslog | grep 'garage_controller' | awk -f /home/pi/parse.awk > /home/pi/garage-door-controller/www/garage_door.html`

Make this file executable:

`chmod +x command`

Create the file 'parse.awk' in /home/pi with your text editor.  It will contain the following:

`BEGIN {print "<html><link href='/css/txtstyle.css' rel='stylesheet' type='text/css' /><p>"}`
`! /toggled/ {print substr($0,1,16) substr($0,45) "<br>"}`
`END {print "</p></html>"}`

Create the file 'txtstyle.css' in /home/pi/garage-door-controller/www/css.  It will contain the following:

`html, body {font-family:Helvetica, Arial, sans-serif; line-height: 10px; white-space: pre-wrap;}`

Edit the 'index.html' file in /home/pi/garage-door-controller/www with your text editor.  Below these lines:

`        <div class="content-primary">`
`          <ul id="doorlist" data-role="listview"></ul>`
`        </div>`

Insert this line:
`        <div id="list"><p><iframe src="garage_door.html" width=400 height=600 frameborder=0 ></iframe></p></div>`

Now set the command to be automatically executed periodically.  You'll be editing the system crontab, so be careful not to change things already there.  You're going to add a line to the file that will execute the command to populate garage_door.html ever 10 minutes.  With your text editor do the following:

`sudo nano /etc/crontab`

Add this line:

`*/10 * * * * root /home/pi/command`

Save and exit nano (or whatever text editor you're using).
 
Notes:
  The files 'txtstyle.css' and 'garage_door.html' must be owned by the pi user in order to work properly. If root owns them, you'll get missing file errors in the web browser.

Every 10 minutes, the system will execute /home/pi/command.  The command pulls every line containing "garage_controller" and passes them to the awk script which converts the text to some minimal html. The script only passes along the lines that don't contain the word "toggled", and pulls out the timestamp and the relevant text.  The output of the script is written to the garage_door.html file in the www folder.  Whenever you browse to the web page, index.html will pull in the text in the garage_door.html file and display it on the page.

Voila!

Gilles94500, 2019-02-16:
Thanks a lot. 

It works perfectly.

 

My 2 cents : the file ‘command’ : I have used ‘tac’ instead of ‘cat’ because I prefer to have the last records first :)

Gilles

NCAvian, 2019-02-16:
Good deal! It was easier than I thought. I'm not an html guru, but I can Google them!Bob