andrewshilliday/garage-door-controller #103: Obtaining code by SVN not working

somerandomguy03, 2019-01-12:
When I enter the following into terminal:

`svn co https://github.com/andrewshilliday/garage-door-controller/trunk ~pi/garage-door-controller`

I get:

`bash: svn: command not found`

Any ideas why?  Did GitHub change something?

andrewshilliday, 2019-01-18:
That error indicates that you do not have subversion installed on your
local computer.

somerandomguy03, 2019-01-19:
Indeed, subversion was not installing properly.  Kept getting answers.  Decided to pull the SD card, completely wipe it and start over.  Everything is working great now.  We had a few power surges that seemed to corrupt the OS, which is why I was attempting to reinstall your app in the first place.  Seems the card was in worse shape than I thought.  All is working now!  Thanks, again, for sharing this great project!

closed 2019-01-22 by andrewshilliday