andrewshilliday/garage-door-controller #104: Alternate configuration for magnetic switch

JBenzie, 2019-01-21:
This isn't so much an issue as a question. I apologize if this isn't the appropriate venue for asking a question, but I couldn't find anywhere else to do so.

Is it possible to have the software recognize the door as open when the magnetic switch is closed, rather than open? Basically, I want to mount the switch in such a way that when the door is opened, the two pieces of the magnetic switch are close to one another and when the door is closed, the switch parts will be separated.

Is this possible via a configuration file or something?

Thanks in advance.
jerry

Gilles94500, 2019-01-22:
Hi Jerry

Just order a different reed switch. Based on the picture below, choose the  ‘NO’ one.

 

 

https://www.aliexpress.com/item/Home-security-alarm-Wired-Window-Magnetic-door-Contact-Sensor-Detector-Switch-for-GSM-reed-switch-door/32888436337.html?spm=2114.search0104.3.1.67cc2ebcwn8uAl&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_319_10892_317_10696_453_10084_454_10083_433_10618_431_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_100031_321_322_10103,searchweb201603_6,ppcSwitch_0&algo_expid=465bf081-a116-4626-8858-ae243591921e-0&algo_pvid=465bf081-a116-4626-8858-ae243591921e

 

Regards,

Gilles

JBenzie, 2019-01-22:
Simple enough. Thank you!

closed 2019-01-22 by JBenzie

andrewshilliday, 2019-01-22:
Hi, before you go through the trouble of using alternate hardware, there is a configuration setting you may want to try playing with.  I've never tested it (as this is not how I've ever configured my doors), but there is a setting in the config.json file for switching the "closed" value of the sensor relay.  The settings are "state_pin_closed_value".  Try changing that from a 0 to a 1 and see if that does the trick.

JBenzie, 2019-01-22:
Thank you. I did see that setting in the config.json file but didn't want to make any assumptions and figured I'd ask first. I don't have anything installed yet, I'm still waiting on the 2-channel relay module to be delivered. I have a NC magnetic reed switch that I'll try first along with that config setting and see how it goes. It's easy enough to change if it doesn't work.

I appreciate the responses. I'll report back once I get things up and running.

Gilles94500, 2019-01-22:
Nice feature. 

I know now that I am running an older version. I don’t have this option in the config.json file :)

Gilles