andrewshilliday/garage-door-controller #106: SSL & Let's Encrypt

NCAvian, 2019-02-10:
I updated the software to get the SSL capability, but was never able to generate a certificate that Chrome on Android was happy with. I think it might have worked if I'd changed the setting to accept all SSL certificates, but that defeats the purpose of using SSL in the first place.

So, I decided to try certificates from Let's Encrypt. The good news is that they're free, but the bad news is that you have to renew them every 90 days.  My initial source of information was here: https://pimylifeup.com/raspberry-pi-ssl-lets-encrypt/

First, edit config.json and set use_https to false.

Since I'm running Raspbian Stretch on the Pi Zero that's running the software, I was able to start with step 5.  I installed the CertBot software using the "Everything Else" step.

In step 6, I port forwarded port 80 from the router to port 8081 and port 443 to port 8444. I have a domain name set up with DynDns so that I can access the server from outside my LAN.  You need to have a domain name set up for the CertBot software to work.

Since we're not using Apache here, I used step 8b with the following command:

sudo certbot certonly --webroot -w /home/pi/garage-door-controller/www -d mydomain.homeip.net

You'll enter a few things the first time you run this, but afterwards, the certificates should renew without further input from you.  When the first certificate is generated, you should get output such as:

- Congratulations! Your certificate and chain have been saved at:
   /etc/letsencrypt/live/mydomain.homeip.net/fullchain.pem
   Your key file has been saved at:
   /etc/letsencrypt/live/mydomain.homeip.net/privkey.pem
   Your cert will expire on 2019-05-11. To obtain a new or tweaked
   version of this certificate in the future, simply run certbot
   again. To non-interactively renew *all* of your certificates, run
   "certbot renew"

Now, edit the config.json file again, setting the use_https flag to true, and add the certificate and key paths as appropriate in the site section of the configuration file.

Updating:
Before updating the certificates, make sure use_https is set to false and port 80 is still forwarded to your Pi. CertBot will fail if it cannot connect port 80 to port 8081 on the Pi.

To test updating the certificates without actually doing it, use this command:

sudo certbot renew --dry-run

To actually renew the certificates, use this command:

sudo certbot renew

And then turn use_https back on in the config.json file.

NCAvian, 2019-02-10:
Some further notes:

I don't believe it actually uses port 443 during the process of generating the certificate. You'll need it forwarded to the Pi to port 8444 to use https, however.

Installing certbot also installs a cron job to run the certificate renewal process twice per day. Certbot actually only renews the certificates about 30 days before they expire.  Since port 80 won't be available when you set use_https = true in the configuration file, you might as well move the file.  As root, move /etc/cron.d/certbot somewhere else, or delete it.  If port 80 were always available on the web server we're using here, you could leave it in place.

I actually have two web servers running on different machines, and access them using port forwarding. I run an Apache server that has port 80 available on the other server and generate SSL certificates for it as well.  If you're doing something like this, you can generate the certificates on a PC, for example, and copy the resulting certificates to the Pi - instead of generating them on the Pi as well.  I did this, and set up a cron job on the server to copy the certificates to the Pi using secure copy.

commputethis, 2020-12-16:
Thinking out loud here.  Could you add a cron job to switch the configuration file to port 80 prior to the cron job for the certificate renewal and then another to switch it back afterwards so it is fully automated?

NCAvian, 2020-12-16:
You could. I have another server where I get credentials, and I have a 
cron job on that server to copy them to the garage door controller Pi.

I actually copy the key file on a weekly basis and just let the server 
update the key whenever it feels a need to do so.

Links:
------
[1] 
https://github.com/andrewshilliday/garage-door-controller/issues/106#issuecomment-745701715
[2] 
https://github.com/

commputethis, 2020-12-16:
I like that option much better.